intel/zephyr.js #920: [docs] Docs of ashell need to be updated

qiaojingx, 2017-03-28:
### Description

Docs of ashell need to be updated
 
### Test Code

[ashell.md](https://github.com/01org/zephyr.js/blob/master/docs/ashell.md)
 
### Actual Result

'Compilation' and 'Connect using serial console' are still using the old ones.

Compilation
[image]

Connect using serial console 
[image]

### Expected Result

This command will enable the shell compilation with:  
`make ide` (builds ashell for the IDE)
`make ashell` (builds for the command line)

### Test Builds

Branch | Commit Id | Test Date | Result
---|---|---|---
master | 9e666f4 | Mar 28, 2017 | Fail
 
### Additional Information

brianjjones, 2017-03-28:
This had been updated.

qiaojingx, 2017-03-29:
@brianjjones , I think 'Connect using serial console' section also need to update. We don't need to `Set 'CONFIG_USB_CDC_ACM=y' in fragments/prj.conf.arduino_101_dev to enable USB CDC ACM device class driver.` now.

brianjjones, 2017-03-29:
@qiaojingx good catch I'll change that as well

brianjjones, 2017-04-05:
Update can be seen here https://github.com/01org/zephyr.js/pull/969

qiaojingx, 2017-04-07:
Verified with commit f8f9621. Closed as fixed.

closed 2017-04-07 by qiaojingx