intel/zephyr.js #924: USAGE FAULT when running HelloWorld.js

makerdiary, 2017-03-28:
Hi guys,
I am trying to running zephyr.js on nRF52832. I can compile the sample  `HelloWorld.js` now, but when I run it on the hardware, a fault came out. Any ideas about this error?
```
***** USAGE FAULT *****
  Executing thread ID (thread): 0x20006aec
  Faulting instruction address:  0x6bfa
  Unaligned memory access
Fatal fault in essential thread! Spinning...
```

Thanks for your reply.
Zelin

jprestwo, 2017-03-28:
We have not tested on the nRF52832 but we are always trying to get new Zephyr boards supported. How did you build ZJS? (e.g. make BOARD=nRF52832). Have you tried building and running a Zephyr sample on the nRF52832?

jprestwo, 2017-03-28:
I tried running hello world on the nRF52-PCA10040 and was able to run it. I followed directions from the Zephyr board support site to setup (download tools):
https://www.zephyrproject.org/doc/boards/arm/nrf52_pca10040/doc/nrf52_pca10040.html

And then built by running:
{code}
make BOARD=nrf52_pca10040 JS=samples/HelloWorld.js
{code}

Then flashed:
{code}
nrfjprog --eraseall -f nrf52
nrfjprog --program outdir/nrf52_pca10040/zephyr.hex -f nrf52
{code}

makerdiary, 2017-03-29:
@jprestwo Thanks for your quick response.
I just tried to build it for nrf52_pca10040, the same fault came out.
I can build the zephyr samples, and they are running fine.

jprestwo, 2017-03-29:
Just to make sure we are talking about the same board, is this the one you have:
http://www.digikey.com/product-detail/en/NRF51-DK/1490-1038-ND/5022449?WT.mc_id=IQ_7595_G_pla5022449&wt.srch=1&wt.medium=cpc&WT.srch=1&gclid=COz7t_CV_NICFUlNfgodiJYMqQ

makerdiary, 2017-03-30:
@jprestwo I am using [nRF52 DK](http://www.nordicsemi.com/eng/Products/Bluetooth-low-energy/nRF52-DK) not nRF51 DK.

grgustaf, 2017-03-30:
That's really what @jprestwo meant; that's what we have here too, nrF52 DK.

grgustaf, 2017-03-30:
@jprestwo hmm it does say unaligned memory access, and we were just talking about the fact that malloc seems to return unaligned. Maybe that makes the nRF52 upset. There may be a config we need to force alignment.

grgustaf, 2017-03-30:
This [nRF52 info](http://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.nrf52%2Fdita%2Fnrf52%2Fmigration%2Ffunctional.html) (see Unlaigned read/write section) sounds like Cortex-M4F can support unaligned access, but if an UNALIGN_TRP bit is set you get this USAGE error. So we could perhaps turn that off, but better would be to make our memory accesses aligned to improve performance, I would think.

grgustaf, 2017-03-30:
@makerdiary, sorry for the flood of comments, but if you'd want to do an experiment you could check out ZJS commit 02d9471, before a recent update to use newlib malloc. Then make sure to run "make update" to go back to whatever Zephyr build that was based on. Then try building again. I suspect that might work, which would be a useful data point.

makerdiary, 2017-04-01:
@grgustaf Thanks for help. I will test it as soon as possible.

jimmy-huang, 2017-08-18:
@makerdiary I have verified that it works on the nRF52 DK (nrf52_pca10040), see #1479 for updated instructions on how to build and flash it.  We don't have nRF52832 but it should also work as long as Zephyr supports it.

makerdiary, 2017-08-22:
@jimmy-huang Awesome, it works on the nRF52832-based board.

qiaojingx, 2018-01-03:
Verified with commit b32969f on nRF52840-pca10056 board. This issue is not reproduced.

closed 2018-01-08 by grgustaf