intel/zephyr.js #936: ashell has parser disabled now

grgustaf, 2017-03-29:
Probably some recent Makefile change messed this up and left SNAPSHOT=on or something?

grgustaf, 2017-03-30:
I should have reported what I actually saw. After building ashell, flashing, and booting, it reported an error about parser not being available or something on the secondary shell (USB0?). Then on the ACM0 if I tried to do an eval or something it would give me an error that there was no parser.

jimmy-huang, 2017-03-30:
@grgustaf works for me, if you can reproduce this, show the steps how to reproduce it, or close this issue.

qiaojingx, 2017-04-01:
Verified with commit f3d587d. This issue can be reproduced.
Steps1: 
1. `make ide`
2. `make dfu`
3. connect ashell with webide(you may met [issue104](https://github.com/01org/zephyrjs-ide/issues/104))
4. Run examples/[Blink.js](https://github.com/01org/zephyr.js/blob/master/samples/arduino/basics/Blink.js) in it.

Error got:
[image]

Steps2: 
1. `make ashell`
2. `make dfu`
3. connect ashell with terminal via `screen /dev/ttyACM0 115200`
4. `acm > load test.js`
5. Input samples/[Blink.js](https://github.com/01org/zephyr.js/blob/master/samples/arduino/basics/Blink.js)  and CTRL+Z to saving
6. `acm > run test.js`

Error info as below:
ACM0:
[image]
USB0:
[image]

grgustaf, 2017-04-04:
Well, I can't reproduce it now on master or on the commit @qiaojingx listed. But yes this is exactly what I saw, I'm glad I wasn't crazy. :)

Apparently we haven't identified the exact steps to reproduce it. I'm lowering to P3 until we can find a consistent way to reproduce it.

jimmy-huang, 2017-04-05:
Looking at the makefile, it could be caused by if you first build with snapshot, then somehow you did something where you removed .snapshot.last_build file, so it mistakenly thinks that jerryscript already have parser enabled, and doesn't rebuild it,

So one way (maybe the only way) to reproduce this is:
1. make JS=Helloworld.js (or anything that uses SNAPSHOT=on)
2. make clean BOARD=linux
3. make ide

Does these steps sound like what you guys might haven done to reproduce this?

jimmy-huang, 2017-04-05:
Submitted possible fix here #974

grgustaf, 2017-04-06:
Submitted another possible fix in #986

commit 7080def7 (2017-04-06): [build] Maintain both snapshot and parser versions of JrS library (#986)

Instead of replacing the same libjerry-core.a with a different version
depending on snapshot vs. parser build, maintain both with different
filenames. Then we don't have to rebuild JerryScript just because we
switched between those two types of builds, and don't need the snapshot
last_build file and related dangers of inconsistency.

Fixes #936
closed 2017-04-06 by jimmy-huang