intel/zephyr.js #945: UDP does not work on Arduino 101

haoxli, 2017-03-30:
# Description
UDP server does not work on Arduino 101

# Test Code
[UDPEchoServ4.js](https://github.com/01org/zephyr.js/blob/master/samples/UDPEchoServ4.js)
[UDPEchoServ6.js](https://github.com/01org/zephyr.js/blob/master/samples/UDPEchoServ6.js)

# Steps to Reproduction
1. `make JS=samples/UDPEchoServ4.js`
2. `make dfu`
3. In another terminal: `echo ‘hello’ | nc -u 192.0.2.1 33333`

# Actual Result
[image]

# Expected Result
No messy code and receive ‘hello’ string.

# Test Builds
Branch | Build ID | Target Device| Date | Result
---|---|---|---|---
master | a2ce873| Arduino 101 | Mar 30, 2017 | Fail

haoxli, 2017-03-30:
Sakari suggested this P1 issue should work in 0.3 release for A101.

pfalcon, 2017-03-30:
Please see https://github.com/01org/zephyr.js/issues/918#issuecomment-290346299 for likely underlying cause and my ideas how to go about resolving such issues consistently.

grgustaf, 2017-04-04:
So the 'garbage' is gone and now the problem is just network configuration like @pfalcon said. I suggested we start by just printing out the IP address on startup and then we'll learn whether it's fixed or randomized, and be able to get send a packet to the device presumably.

jprestwo, 2017-04-05:
With my recent change UDP, the A101 should work. The key is to add a new ip route for the bt0 interface after you have connected to the A101 using the OCF BLE method:
```
ip -6 route add 2001:db8::/64 dev bt0
```

After this you can then ping `2001:db8::1` using nc:
```
echo "hello" | nc -u 2001:db8::1 33333
```
You should then see the echo back with the first character missing (i.e. "ello")

qiaojingx, 2017-04-06:
Verified with commit ddcbbbe on arduino. UDP6 is working after taking @jprestwo 's steps.
UPD4 is still not working. 
`ip route add 192.0.2.0/24 dev bt0`
`echo "hello" | nc -u 192.0.2.1 33333 `

jprestwo, 2017-04-06:
@qiaojingx, I would not expect IPv4 to work on the Arduino since its using 6lowpan, which is IPv6.

haoxli, 2017-04-07:
@jprestwo thanks. So we will only check IPv6 on Arduino and close the issue as fixed.

closed 2017-04-07 by haoxli