intel/zephyr.js #956: Enable accelerometer and magnetometer sensors on K64F

grgustaf, 2017-03-31:


jimmy-huang, 2017-04-01:
Yeah, no magnetometer yet, but shouldn't be difficult to add

kenchris, 2017-04-14:
Yes, the spec covers magnetometers

qiaojingx, 2017-06-28:
Accelerometer sensor is enabled on K64F. Verified with samples([FXOS8700Accelerometer.js)](https://github.com/01org/zephyr.js/blob/master/samples/FXOS8700Accelerometer.js). Samples of magnetometer sensor are not added yet.
Commit: 8a95b21+ #1312 
Platform: FRDM-K64F

qiaojingx, 2017-06-30:
Accelerometer and Magnetometer sensors are enabled on FRDM-K64F. Verified with samples([FXOS8700Accelerometer.js](https://github.com/01org/zephyr.js/blob/master/samples/FXOS8700Accelerometer.js) and [FXOS8700Magnetometer.js](https://github.com/01org/zephyr.js/blob/master/samples/FXOS8700Magnetometer.js)). One known issue is for Magnetometer sensor(#1327).
Commit: c2ca034
Platform: FRDM-K64F

qiaojingx, 2017-07-07:
Accelerometer and Magnetometer sensors are enabled on FRDM-K64F. Verified with samples([FXOS8700Accelerometer.js](https://github.com/01org/zephyr.js/blob/master/samples/FXOS8700Accelerometer.js) and [FXOS8700Magnetometer.js](https://github.com/01org/zephyr.js/blob/master/samples/FXOS8700Magnetometer.js)). One known issue is for Magnetometer sensor(#1327).
Commit ID: 0334065
Platform: FRDM-K64F.
ZephyrSDK: 0.9.1
Zephyr:1.8

jimmy-huang, 2017-08-02:
Closing this as it is implemented.

closed 2017-08-02 by jimmy-huang