intel/zephyr.js #970: [buffer] Cannot work for buffer on Linux module

cuiyanx, 2017-04-05:
### Description

Cannot work for buffer on Linux module. 

Console log:
`Segmentation fault (core dumped)`
 

### Test Code

[/samples/Buffer.js](https://github.com/01org/zephyr.js/blob/master/samples/Buffer.js)
[/samples/tests/BufferMemory.js](https://github.com/01org/zephyr.js/blob/master/samples/tests/BufferMemory.js)
[/tests/test-buffer.js](https://github.com/01org/zephyr.js/blob/master/tests/test-buffer.js)
[/tests/test-buffer-rw.js](https://github.com/01org/zephyr.js/blob/master/tests/test-buffer-rw.js)

 

### Steps to Reproduction

 

### Actual Result

Throw out error and cannot work.

 

### Expected Result

No error and work well.
 

### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | ddcbbb| Linux | April 5, 2017 | Fail

 

### Additional Information

grgustaf, 2017-04-05:
@cuiyanx, the error it throws is working according to [your request](https://github.com/01org/zephyr.js/issues/916#issuecomment-290284288) in #946. :) However, I've submitted #966 which should be merged today, which fixes the test to consider that error a "pass".

grgustaf, 2017-04-05:
You listed all those four tests, but I think it was probably only test-buffer.js that failed? That works w/ #966. The other also work for me, on all three platforms: a101, k64f, and linux; but I expect they did before too.

If that's true, it would be nice if you could make your report more specific to the failing test.

cuiyanx, 2017-04-06:
[image]

cuiyanx, 2017-04-06:
On K64:
[image]
[image]

qiaojingx, 2017-04-06:
Verified with commit ddcbbbe + #966 on K64F and Linux. This issue is still reproducible.

haoxli, 2017-04-06:
This should be a release block issue which block Buffer API on K64F and Linux.

qiaojingx, 2017-04-06:
This issue is probably caused by #964.
commit 6438700 failed
commit 7bf8e0b pass

jimmy-huang, 2017-04-07:
I can reproduce this, it's the same bug as #988, buffer fails on create

grgustaf, 2017-04-07:
Fixed by #989

closed 2017-04-07 by jimmy-huang

qiaojingx, 2017-04-07:
Verified with commit f8f9621 on K64f and Linux. This issue is not reproducible.

cuiyanx, 2018-01-10:
Fixed and closed

PR #989 (2017-04-07): [buffer] Check for NULL return buffer pointers in create func
Fixes #970 and #988.
commit f8f96210 (2017-04-07): [buffer] Check for NULL return buffer pointers in create func (#989)

Fixes #970 and #988.