intel/zephyr.js #971: [K64-pins] D14 and D15 can be used as GPIO outputs

cuiyanx, 2017-04-05:
### Description

[/docs/k64f_pins.md](https://github.com/01org/zephyr.js/blob/master/docs/k64f_pins.md):
`D14 and D15 can be used as GPIO inputs but not outputs currently.`

But there are all failed when I test [/tests/test-k64f-pins.js](https://github.com/01org/zephyr.js/blob/master/tests/test-k64f-pins.js).
 

### Test Code

[/tests/test-k64f-pins.js](https://github.com/01org/zephyr.js/blob/master/tests/test-k64f-pins.js)

 

### Steps to Reproduction

 

### Actual Result

D14 and D15 can be used as GPIO outputs.
 

### Expected Result

No fail and can not be used as GPIO outputs.
 

### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | ddcbbb | K64 | April 5, 2017 | Fail

 

### Additional Information

grgustaf, 2017-04-05:
This is not a bad thing. :) Back when we first implemented the GPIO support we didn't know how to make D14 and D15 work as outputs, and didn't know whether it could be done.

The test should be updated.

grgustaf, 2017-04-05:
Hmm maybe I didn't understand. Are you saying *none8 of the pins work as outputs? I guess I will test again and update the document you mentioned.

cuiyanx, 2017-04-06:
@grgustaf If D14 and D15 can be used as GPIO outputs, please update the [/docs/k64f_pins.md](https://github.com/01org/zephyr.js/blob/master/docs/k64f_pins.md); If not, please make D14 and D15 do not work as output pins. I think that function and documentation should be unified.

grgustaf, 2017-04-06:
Yes, but you should understand when the docs say something "doesn't work" that doesn't mean it "shouldn't work". If it suddenly starts working, that's not a bad thing. But we should update the documentation. :)

grgustaf, 2017-04-06:
Submitted doc change as PR #979.

grgustaf, 2017-04-07:
Reconsidering as documentation bug; removing 'invalid' label.

qiaojingx, 2017-04-07:
Verified with commit f8f9621 on k64f. [k64f_pins.md](https://github.com/01org/zephyr.js/blob/master/docs/k64f_pins.md) has been updated.

cuiyanx, 2017-04-07:
Yes, fixed and closed

closed 2017-04-07 by cuiyanx