intel/zephyr.js #977: [grovelcd] Grovelcd does not work on K64

cuiyanx, 2017-04-06:
### Description

Grovelcd does not work on K64, no console log.
 

### Test Code

[/samples/I2C.js](https://github.com/01org/zephyr.js/blob/master/samples/I2C.js)
[/samples/GroveLCD.js](https://github.com/01org/zephyr.js/blob/master/samples/GroveLCD.js)
[/tests/test-i2c.js](https://github.com/01org/zephyr.js/blob/master/tests/test-i2c.js)
[/tests/test-grovelcd.js](https://github.com/01org/zephyr.js/blob/master/tests/test-grovelcd.js)
[/tests/test-grovelcd-rgb-manual.js](https://github.com/01org/zephyr.js/blob/master/tests/test-grovelcd-rgb-manual.js)
[/tests/test-grovelcd-ds-manual.js](https://github.com/01org/zephyr.js/blob/master/tests/test-grovelcd-ds-manual.js)
[/tests/test-grovelcd-cursor-manual.js](https://github.com/01org/zephyr.js/blob/master/tests/test-grovelcd-cursor-manual.js)

 

### Steps to Reproduction

 

### Actual Result

Does not work and no console log.
 

### Expected Result

Work well.
 

### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | ddcbbb| K64 | April 6, 2017 | Fail

 

### Additional Information

[/tests/test-i2c.js](https://github.com/01org/zephyr.js/blob/master/tests/test-i2c.js):
[image]

[/samples/I2C.js](https://github.com/01org/zephyr.js/blob/master/samples/I2C.js):
[image]

qiaojingx, 2017-04-06:
This issue is probably caused by #954.
commit 9e12add failed
commit ed04317 pass

grgustaf, 2017-04-06:
Yeah, that makes sense. We probably need to identify whether I2C is being used and if so, not pinmux those pins to GPIO.

closed 2017-04-06 by grgustaf

qiaojingx, 2017-04-07:
Verified with commit f8f9621 on k64f. This issue is not reproducible.

cuiyanx, 2018-01-10:
Fixed and closed

PR #987 (2017-04-06): [i2c] Fix bug where pinmux is screwing up I2C
Fixes #977
commit fb41b478 (2017-04-06): [i2c] Fix bug where pinmux is screwing up I2C (#987)

Fixes #977