intel/zephyr.js #978: [udp] Can't receive message on FRDM-K64F

qiaojingx, 2017-04-06:
### Description

 cannot receive the message sent by terminal.

### Test Code

[UDPEchoServ4.js](https://github.com/01org/zephyr.js/blob/master/samples/UDPEchoServ4.js);[UDPEchoServ6.js](https://github.com/01org/zephyr.js/blob/master/samples/UDPEchoServ6.js)
 
### Steps to Reproduction

1. `make BOARD=frdm_k64f JS=samples/UDPEchoServ4.js`
2. cp outdir/frdm_k64f/zephyr.bin /media/<USERNAME>/MBED/
3. In another terminal: echo ‘hello’ | nc -u 192.0.2.1 33333

### Actual Result

 No message is received.

### Expected Result

receive ‘hello’ string.
 
### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | ddcbbbe | FRDM-K64F | Apr 6, 2017 | Fail
 
### Additional Information

grgustaf, 2017-04-07:
I believe @jprestwo said this is addressed by #980.

qiaojingx, 2017-04-07:
Verified with commit f8f9621 on K64f. Message can be received when running UDPEchoServ4.js.
Will verify UDPEchoServ6.js later because of my router problem.

qiaojingx, 2017-04-11:
Verified with commit cc2311a on K64F. Message can be received when running UDPEchoServ4.js and UDPEchoServ6.js. Closed.

closed 2017-04-11 by qiaojingx