intel/zephyr.js #990: [k64f] D5 pin is working abnormally on K64f

cuiyanx, 2017-04-07:
### Description

As input pins, D5 can not recognise low voltage as 'false'; 

Pin | Input Voltage(V) | Read Value
---- | ---- | ----
D5 | 4.79 | true
D5 | 0 | null
D4 | 4.79 | true
D4 | 0 | false

As output pin, output voltage is too high.

Pin | Write Value | Output Voltage(V)
---- | ---- | ----
D5 | true | 3.241
D5 | false | 2.457
D4 | true | 3.219
D4 | false | 1.043
 

### Test Code

[/samples/tests/GPIOInputs-k64f.js](https://github.com/01org/zephyr.js/blob/master/samples/tests/GPIOInputs-k64f.js)
[/samples/tests/GPIOOutputs-k64f.js](https://github.com/01org/zephyr.js/blob/master/samples/tests/GPIOOutputs-k64f.js)

 

### Steps to Reproduction

Run test case and D5 no change event.
 

### Actual Result

D5 no change event on test case.
 

### Expected Result

Work well as GPIO pin.
 

### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | f8f96210| K64f | April 7, 2017 | Fail

 

### Additional Information

grgustaf, 2017-04-07:
That's a cool bug, very interesting. :)

grgustaf, 2017-04-07:
This could turn out to be something we're impacting, with pinmux values in an invalid state and muliple onboard devices driving to the same pin or something like that. But it's also not clear that it's our fault so I'm making it lower priority for now.