intel/zephyr.js #1007: [docs] Wrong links in console.md and ocf.md,

qiaojingx, 2017-04-11:
### Description

Wrong links in console.md and ocf.md
 
### Test Code

[console.md](https://github.com/01org/zephyr.js/blob/master/docs/console.md); [ocf.md](https://github.com/01org/zephyr.js/blob/master/docs/ocf.md)

### Steps to Reproduction

1. Click [Console sample](https://github.com/01org/zephyr.js/blob/master/samples/Console.js) in [Sample Apps](https://github.com/01org/zephyr.js/blob/master/docs/console.md#sample-apps)

2. Click [OCF Sensor Client](https://github.com/01org/zephyr.js/blob/master/samples/OcfSensorClient.js) in [Client Samples](https://github.com/01org/zephyr.js/blob/master/docs/ocf.md#client-samples)

### Actual Result

1. 404 page not found
2. 404 page not found

### Expected Result

1. Console.js should be opened.
    The right link address should be `/samples/tests/Console.js`, not `/samples/Console.js`
2. Can't find `OcfSensorClient.js`. Please add `OcfSensorClient.js`.
 
### Test Builds

Branch | Commit Id | Test Date | Result
---|---|---|---
master | 1e9da6f | Apr 11, 2017 | Fail
zjs-0.3 | f8f9621 | Apr 11, 2017 | Fail
 
### Additional Information

closed 2017-04-11 by grgustaf

qiaojingx, 2017-04-12:
Verified with commit 2487b22 in master. Issue was fixed.
But it is still reproduced in zjs-0.3. 
@jprestwo Can we also merge #1012 to zjs-0.3 branch ?

cuiyanx, 2017-04-19:
Sync 0.3 docs fix and sample added: #1036

qiaojingx, 2017-04-20:
Verified with commit 46c8cc2 in zjs-0.3 branch. This issue is not reproducible.

PR #1012 (2017-04-11): [docs/ocf] Fix console doc link and add sample for OCF
A sample was referenced in OCF docs but never added

Fixes #1007
commit 2487b22c (2017-04-11): [docs/ocf] Fix console doc link and add sample for OCF (#1012)

A sample was referenced in OCF docs but never added

Fixes #1007