intel/zephyr.js #1008: [docs] Inaccurate commands in README.md

qiaojingx, 2017-04-11:
### Description

Inaccurate commands in README.md
 
### Test Code

[README.md](https://github.com/01org/zephyr.js/blob/master/README.md)
 
### Steps to Reproduction

1. Follow the introduction[ FRDM-K64F Platform](https://github.com/01org/zephyr.js/#frdm-k64f-platform)
Try to build the Zephyr OS hello_world sample to narrow down the problem:
[image]
 
2. Follow the introduction [Networking with QEMU](https://github.com/01org/zephyr.js/#networking-with-qemu) to setup net-tools
Terminal 1:
`sudo ./net-tools/loop-socat.sh`
Terminal 2:
`./net-tools/loop-slip-tap.sh`

### Actual Result

1. `bash: cd: deps/zephyr/samples/hello_world/nanokernel: No such file or directory`
2.  
[image]

### Expected Result

1. The path should be `cd deps/zephyr/samples/hello_world`
2. Terminal 1:
` ./loop-socat.sh`
Terminal 2:
`sudo ./loop-slip-tap.sh`

### Test Builds

Branch | Commit Id | Test Date | Result
---|---|---|---
master | 1e9da6f | Apr 11, 2017 | Fail
zjs-0.3 | f8f9621 | Apr 11, 2017 | Fail
 
### Additional Information

closed 2017-04-11 by jimmy-huang

qiaojingx, 2017-04-12:
@jprestwo K64 build instruction has been updated in commit 2487b22. But the instruction of [Networking with QEMU](https://github.com/01org/zephyr.js/#networking-with-qemu) is still not inaccurate. Please find current behavior from the picture in description. I think it is better to update as below(keep consistent with https://www.zephyrproject.org/doc/subsystems/networking/qemu_setup.html):
Terminal 1:
`./loop-socat.sh`
Terminal 2:
`sudo ./loop-slip-tap.sh`

cuiyanx, 2017-04-19:
Sync 0.3 docs fix and sample added: #1036

qiaojingx, 2017-04-20:
Verified with commit 46c8cc2 in zjs-0.3 branch. K64 build instructions have been fixed. Other instructions are the same as master.

PR #1011 (2017-04-11): [docs] Fix K64 build instructions
Fixes #1008
commit 9fddd607 (2017-04-11): [docs] Fix K64 build instructions (#1011)

Fixes #1008