intel/zephyr.js #1015: [ashell] gpio.onchange can't be triggered each time

qiaojingx, 2017-04-12:
### Description

gpio.onchange can't be triggered each time

### Test Code

[Button.js](https://github.com/01org/zephyr.js/blob/master/samples/arduino/digital/Button.js); [TwoButtons.js](https://github.com/01org/zephyr.js/blob/master/samples/TwoButtons.js)
 
### Steps to Reproduction

1. `make ashell`
2. `make dfu`
3. Connect ashell with terminal via `minicom -D /dev/ttyACM0`
4. `acm> load test.js` and Enter
5. `RAW> /*Paste code of test cases*/`and CTRL+Z to finish transfer
6. `acm> run test.js`
7. Press button to check onboard led.
 
### Actual Result

1. Red led is lighted by default.
2. State of red led is not changed when press button most of time.
 
### Expected Result

1. Red led should be off by default.
2. Red led should be turned on each time when press button.

### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | 2487b22 | Arduino 101 | Apr 12, 2017 | Fail
 
### Additional Information

This issue is also reproduced when running these samples in WebIDE.

brianjjones, 2017-04-14:
This is a Zephyr bug, https://jira.zephyrproject.org/browse/ZEP-2024

The workaround is to edit fragments/zjs.conf.dev and comment out ZJS_SENSOR.  Long story short, we currently can't use GPIO as an input and sensor at the same time.

qiaojingx, 2017-04-21:
Verified with commit 46c8cc2 against  zjs-0.3 branch. This issue is not reproducible.
Unable to verify with latest commit(dbcd8ea) in master due to reading long native module names failed.

qiaojingx, 2017-04-25:
Verified with commit 93d58fa in master. This issue is not reproducible. Closed

closed 2017-04-25 by qiaojingx