intel/zephyr.js #1024: Fail to compile using GCC ARM on Mac

makerdiary, 2017-04-16:
Hi guys,
When I use GCC ARM on Mac to compile an ARM boards(like nRF52832-based board), something error about the VFP register arguments.

Here is the log, any ideas about that? Thanks.

```
...

~/gcc-arm-none-eabi/gcc-arm-none-eabi-6-2017-q1-update/bin/../lib/gcc/arm-none-eabi/6.3.1/../../../../arm-none-eabi/bin/ld: error: zephyr_prebuilt.elf uses VFP register arguments, ~/gcc-arm-none-eabi/gcc-arm-none-eabi-6-2017-q1-update/bin/../lib/gcc/arm-none-eabi/6.3.1/../../../../arm-none-eabi/lib/libm.a(lib_a-s_nextafter.o) does not
~/gcc-arm-none-eabi/gcc-arm-none-eabi-6-2017-q1-update/bin/../lib/gcc/arm-none-eabi/6.3.1/../../../../arm-none-eabi/bin/ld: failed to merge target specific data of file ~/gcc-arm-none-eabi/gcc-arm-none-eabi-6-2017-q1-update/bin/../lib/gcc/arm-none-eabi/6.3.1/../../../../arm-none-eabi/lib/libm.a(lib_a-s_nextafter.o)
~/gcc-arm-none-eabi/gcc-arm-none-eabi-6-2017-q1-update/bin/../lib/gcc/arm-none-eabi/6.3.1/../../../../arm-none-eabi/bin/ld: error: zephyr_prebuilt.elf uses VFP register arguments, ~/gcc-arm-none-eabi/gcc-arm-none-eabi-6-2017-q1-update/bin/../lib/gcc/arm-none-eabi/6.3.1/../../../../arm-none-eabi/lib/libc.a(lib_a-abort.o) does not
~/gcc-arm-none-eabi/gcc-arm-none-eabi-6-2017-q1-update/bin/../lib/gcc/arm-none-eabi/6.3.1/../../../../arm-none-eabi/bin/ld: failed to merge target specific data of file ~/gcc-arm-none-eabi/gcc-arm-none-eabi-6-2017-q1-update/bin/../lib/gcc/arm-none-eabi/6.3.1/../../../../arm-none-eabi/lib/libc.a(lib_a-abort.o)

...

```

jimmy-huang, 2017-04-21:
there are some issues with the ARM compiler, can you check if this is a Zephyr bug? for example, can you build Zephyr native sample app like helloworld?  If you are running into the same issue with the native app, then it's a Zephyr bug and probably should file it on upstream

makerdiary, 2017-04-22:
@jimmy-huang Zephyr is fine, I can compile the Zephyr samples using GCC-ARM. But I will get the same issue when compiling Jerryscript along. I hava committed a issue about this problem:
https://github.com/jerryscript-project/jerryscript/issues/1684

qiaojingx, 2017-04-26:
I am unable to compile zephyr.js for K64f using gcc-arm on Mac. It prompt that 'fatal error: stdio.h: No such file or directory.'

[image]
 Following https://github.com/01org/zephyr.js#other-targets-like-frdm-k64f-or-possibly-other-arm-boards-on-mac  

Am I missing something?

jimmy-huang, 2017-09-25:
building native on Mac is not really supported at the moment besides just build with BOARD=linux, both arduino_101 and frdm_k64f will require you to install a different compiler since Zephyr doesn't provide the toolchain with the SDK.  We can revisit build support for Mac if this becomes a higher priority feature.

qiaojingx, 2017-12-04:
Verified with latest commit cbec79d in master, frdm_k64f and nrf52_pca10040 can be built on MacOS now.
But a little change needed in [README.md](https://github.com/01org/zephyr.js#other-targets-like-frdm-k64f-or-possibly-other-arm-boards-on-mac), or build will run into error.
change `export GCCARMEMB_TOOLCHAIN_PATH="~/Downloads/gcc-arm-none-eabi-6-2017-q1-update/"`
to `export GCCARMEMB_TOOLCHAIN_PATH="/Users/<USERNAME>/Downloads/gcc-arm-none-eabi-6-2017-q1-update"`

closed 2017-12-07 by jimmy-huang

cuiyanx, 2017-12-08:
Fixed and closed

PR #1692 (2017-12-07): [build] Update README.md to support k64f and nrf52 boards build on MacOS
Fixed #1024