intel/zephyr.js #1025: [modules] Allow requiring the same modules on modules and case at the same time

cuiyanx, 2017-04-18:
### Description

If importing A module to B module and importing A module to C module, I can not import A module to test case after importing B and C module.
 

### Test Code

Import `arduino101_pins` module to Testmodule1.js:
```
function Testmodule1() {
    var testmodule1 = {};
    var pins = require("arduino101_pins");
    var gpio = require("gpio");
    var pinX = gpio.open({ pin: pins.IO4, direction: "in" });
    return testmodule1;
};
module.exports.Testmodule1 = new Testmodule1();
```

Import `arduino101_pins` module to Testmodule2.js:
```
function Testmodule2() {
    var testmodule2 = {};
    var pins = require("arduino101_pins");
    var gpio = require("gpio");
    var pinY = gpio.open({ pin: pins.IO7, direction: "in" });
    return testmodule2;
};
module.exports.Testmodule2 = new Testmodule2();
```
Import `arduino101_pins` module to test case:
```
var gpio = require("gpio");
var pins = require("arduino101_pins");
var testmodule1 = require("Testmodule1.js");
var testmodule2 = require("Testmodule2.js");

console.log("--------1");
var pinA = gpio.open({ pin: pins.IO2, direction: "in" });
console.log("--------2");
```

 

### Steps to Reproduction

1. Import A module to B module.
2. Import A module to C module.
3. Import B and C module to D module.
4. Import A module to D module.
 

### Actual Result

Exit test case.
 
[image]

### Expected Result

Work well.
 

### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | 47e8aef | Arduino 101 | April 18, 2017 | Fail

 

### Additional Information

rwaldron, 2017-04-18:
Node.js manages this by caching the export object of required modules at runtime, and [returning that export object on future calls to require the same module](https://github.com/nodejs/node/blob/a94a5da78c6fdecf1aef541db237a1c96a9801a3/lib/internal/bootstrap_node.js#L479-L487).

grgustaf, 2017-04-18:
Yeah our require() will return a singleton already as well, so not sure where the problem lies exactly. It's probably an issue earlier in the new recursive parsing code.

grgustaf, 2017-04-19:
Hmm, just looking at that I saw what I think is a bug that I fixed in #1033. So it might be worth testing this w/ that PR. I'm not sure I'll get to that tonight, anyone is welcome to try. :)

brianjjones, 2017-04-19:
I'll test when I get in later.  Pretty sure JS modules are fine, but I don't know about native.  I'll try with grgustaf's change too.

brianjjones, 2017-07-26:
This seems to work for me now.  Note that GPIO doesn't use pins.IO2 etc, it no longer uses the arduino101_pins module.

cuiyanx, 2017-08-01:
Fixed and closed

closed 2017-08-01 by cuiyanx