intel/zephyr.js #1026: [callback] Callback server throw out error when callback function called too fast

cuiyanx, 2017-04-18:
### Description

When I use IR Induction sensor on a101, I find that gpio.onchange function is called too fast and callback server throw out error.
 

### Test Code

Test code:
```
console.log("Test case for callback server");

var pins = require("arduino101_pins");
var gpio = require("gpio");

var pinA = gpio.open({
    pin: pins.IO2,
    direction: "in",
    edge: "any"
});

pinA.onchange = function(event) {
    console.log("Senser read value: ", pinA.read());
};

```
 

### Steps to Reproduction

1. Connect sensor to a101.
2. Code test case for reading sensor value.
3. Callback server throw out error.
 

### Actual Result

[image]
[image]

 

### Expected Result

Work well and no error.
 

### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | 47e8aef | Arduino 101 | April 18, 2017 | Fail

 

### Additional Information

grgustaf, 2017-04-19:
I believe the test is invalid. Printing text is a [very slow operation](https://github.com/01org/zephyr.js/wiki/Performance-Experiments), so you're using like 60k clock cycles in your onchange function just for the printing.

If you change your test to instead do something fast like just reading the pin value and recording it in an array, your function will run about 10x faster and I doubt you'll see this problem.

Can you tell us what sensor you're using?

cuiyanx, 2017-04-24:
@grgustaf As I guess and thanks to know. So, be closed. And this is an [IR Induction sensor](https://item.taobao.com/item.htm?spm=a1z10.5-c-s.w4002-15635545148.10.3Y02u0&id=15584417726).

closed 2017-04-24 by cuiyanx