intel/zephyr.js #1027: [ashell] ashell hangs when execute command 'error'

qiaojingx, 2017-04-18:
### Description

ashell hangs when execute command 'error'

### Test Code

[ashell.md](https://github.com/01org/zephyr.js/blob/master/docs/ashell.md)
 
### Steps to Reproduction

1. `make ashell`
2. `make dfu`
3. `acm> error `

### Actual Result

ashell hangs

### Expected Result

Prints an error using JerryScript
 
### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | 0b11671 | Arduino 101 | Apr 18, 2017 | Fail
 
[image]

### Additional Information

brianjjones, 2017-04-20:
This one also expects you pass it a string.  For example 'error Test!'.  @grgustaf  This seems like a useless function, shall I just remove it?

closed 2017-08-15 by grgustaf

qiaojingx, 2017-08-16:
Verified with commit 212747a.  Command 'error' was removed from ashell.

PR #1465 (2017-08-15): [ashell] Removed error command from Ashell
The error is not a useful command and should be removed as
it is already removed from the help menu, this fixes
a potential lock up issue when used without passing an arugment

Fixes #1027
commit d51c11bc (2017-08-15): [ashell] Removed error command from Ashell (#1465)

The error is not a useful command and should be removed as
it is already removed from the help menu, this fixes
a potential lock up issue when used without passing an arugment

Fixes #1027