intel/zephyr.js #1028: [ashell] 'No enough arguments' returns when execute command 'get'

qiaojingx, 2017-04-18:
### Description

'No enough arguments' returns when execute command 'get'

### Test Code

[ashell.md](https://github.com/01org/zephyr.js/blob/master/docs/ashell.md)
 
### Steps to Reproduction

1. `make ashell`
2. `make dfu`
3. `acm> get`

### Actual Result

'No enough arguments' returns

### Expected Result

Get states on the shell
 
### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | 0b11671 | Arduino 101 | Apr 18, 2017 | Fail
 
[image]

### Additional Information

grgustaf, 2017-04-19:
I'm not sure what this command is supposed to do, but I presume it takes an argument, so it's a matter of better documentation; the error message is probably more or less alright.

brianjjones, 2017-04-20:
It is poorly set up right now, it expects you to call 'get transfer' to get the current transfer mode. It will then print either 'raw' or 'ihex'

@grgustaf Zoltan and Salvatore agree that there's no reason for ihex at all, so this command will likely go the way of the dinosaur soon. I don't plan on fixing it until then, unless you see a reason to.

xiaoyu3x, 2017-10-20:
hi, @qiaojingx 
According to latest [ashell](https://github.com/01org/zephyr.js/blob/master/docs/ashell.md#commands) doc, 'error' cmd has been removed, so closed it please.

qiaojingx, 2017-10-26:
Close since this command has been removed from [ashell.md](https://github.com/01org/zephyr.js/blob/master/docs/ashell.md).

closed 2017-10-26 by qiaojingx