intel/zephyr.js #1029: [ashell] callback server not stop while callback function wrote off after execute 'stop'

qiaojingx, 2017-04-18:
### Description

callback server not stop while callback function wrote off after execute 'stop'

### Test Code

[ashell.md](https://github.com/01org/zephyr.js/blob/master/docs/ashell.md)
 
### Steps to Reproduction

1. `make ashell`
2. `make dfu`
3. `acm> load test.js` and Enter
4. `RAW> /*Paste code of BMI160Accelerometer.js */`and CTRL+Z to finish transfer
5. `acm> run test.js`
6. `acm> stop`

### Actual Result

callback server not stop while callback function wrote off after execute 'stop'
[image]

### Expected Result

Sample stops running completely.
 
### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | 0b11671 | Arduino 101 | Apr 18, 2017 | Fail
 
### Additional Information

grgustaf, 2017-04-19:
FYI we've now disabled sensor code from ashell by default, but you can reenable it manually in fragments/zjs.conf.dev

brianjjones, 2017-04-20:
Currently I was not checking where in the cycle the main loop was when stopping ashell.  This means its possible for ashell to clear everything out while mainloop is trying to call a callback, causing the issue you see. Certainly a bug, working on a fix.

closed 2017-05-16 by grgustaf

qiaojingx, 2017-05-17:
Verified with commit 6a27658. This issue is not reproducible.

PR #1166 (2017-05-16): [sensor] Fix sensor cannot stop/start in IDE/Ashell
There are two bugs in ashell, where we aren't checking
the polling flag correctly on ARC and we are not
stopping the ARC from sending sensor changes when you
stop the JS.  Also fixes bug where we are not cleaning
up the module properly in the cleanup() function.

Fixes #1029