intel/zephyr.js #1052: ashell doesn't work for qemu_x86

pfalcon, 2017-04-23:
So, I finally went to look at ashell, and of course started with the most comfortable way to do so: building it for qemu_x86. That doesn't work at all, and looking further series of issue were identified:

1. Makefile is coded in a way that it's either qemu_x86 or ashell.

2. Makefile and source code lack explicit selection of WebUSB support, and treat USB CDC support not being selected as WebUSB being selected, even if WebUSB support was not requested. But what about the most basic way to communicate with an embedded device - UART?

3. Ashell documentation, https://github.com/01org/zephyr.js/blob/master/docs/ashell.md, describe it as means of interactive prompt for Zephyr.js, but ashell config file actually force particular USB options, particular file system drivers, Bluetooth support and particular options for it, etc.

4. Ashell source code is sprinkled with code snippets specific to Arduino 101.

So, it would be nice to know status/roadmap for ashell. Besides qemu_x86, currently it won't work for frdm_k64f, etc.

pfalcon, 2017-04-23:
To provide a complete bugreport, currently, running `make BOARD=qemu_x86 ashell` fails with:

~~~
  CC      subsys/disk/disk_access_flash.o
/mnt/hdd/projects-3rdparty/Embedded/Zephyr/zephyr.js/deps/zephyr/subsys/disk/disk_access_flash.c:21:30: error: ‘CONFIG_DISK_ERASE_BLOCK_SIZE’ undeclared here (not in a function)
 static uint8_t read_copy_buf[CONFIG_DISK_ERASE_BLOCK_SIZE];
~~~

Though it's not much relevant, because it's not even ashell what's being built, but some more or less random Zephyr config used causes the error.

brianjjones, 2017-04-26:
There isn't anything specific to the A101 in the ashell code other than a call to clean up IPM messages.  However, it does appear that some of the modules that are used / included in ashell may not be supported on these other board options.  For qemu, it looks like it doesn't support the filesystem module.  We'll take a look and see what needs to be added or/and removed to get other platforms working.

pfalcon, 2017-04-27:
> There isn't anything specific to the A101 in the ashell code other than a call to clean up IPM messages. However, it does appear that some of the modules that are used / included in ashell may not be supported on these other board options.

Well, so it would help to separate "ashell" module, which is nothing but an interactive prompt over a serial connection, from all other modules included *with* (not *in*!) ashell. The latter is definitely board specific, and I wouldn't have a better idea how to handle it, except by having multiple "matrix" configs for each of the board. One axis of matrix is a board, another would be (at least): non-networked vs networked vs ashell (which is by definition networked) builds.

pfalcon, 2017-04-27:
> For qemu, it looks like it doesn't support the filesystem module.

I don't see why that would be the case. In Zephyr, FS can use either a flash driver, or ramdisk driver. Ramdisk implementation is quite adhoc (there's a fixed, non-configurable disk size, can you believe that?), but it's there.

When I built for qemu_x86 the blocking culprit actually was ... well, let me put how I see it: incorrect handling of the serial connection. Zephyr.js has the same issue as being fixed by this patch: https://gerrit.zephyrproject.org/r/#/c/12919/ . That's one, another one is that you don't even test return value of uart_fifo_fill(), which you should, and you shouldn't expect it being able to take more than 1 byte at time (and if you don't call it from ISR after TX flag check, you should expect that it make take 0 bytes). With these issues, serial connection somehow works for a101 USB, but immediately breaks down for a real UART (there's an infinite loop). And it works on a101 USB to a degree - I believe infamous "90 bytes" issue might be rooted in not checking the return value of uart_fifo_fill().

brianjjones, 2017-07-06:
This faces the same issue as https://github.com/01org/zephyr.js/issues/1054 , there are no USB drivers to support this.  So for now, its a CANTFIX

grgustaf, 2017-07-26:
I suspect more modularity for ashell is the kind of thing @zolkis addressed in his major revisions to ashell. As I just commented on #1054, the USB drivers only prevent us doing WebUSB with the ide, not ashell itself. So probably we should look at this in the context of "bshell" when we have a chance.

grgustaf, 2018-01-24:
Lowering priority since this would be a development / debugging tool, not a core feature.