intel/zephyr.js #1054: [ashell] build failure for frdm_k64f

JammyZhou, 2017-04-24:
When build ashell or ide for frdm_k64f, errors below happened.

/home/test/zephyr.js/deps/zephyr/subsys/disk/disk_access_flash.c:21:30: error: ‘CONFIG_DISK_ERASE_BLOCK_SIZE’ undeclared here (not in a function)
 static uint8_t read_copy_buf[CONFIG_DISK_ERASE_BLOCK_SIZE];
                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/test/zephyr.js/deps/zephyr/subsys/disk/disk_access_flash.c: In function ‘lba_to_address’:
/home/test/zephyr.js/deps/zephyr/subsys/disk/disk_access_flash.c:35:15: error: ‘CONFIG_DISK_FLASH_START’ undeclared (first use in this function)
  flash_addr = CONFIG_DISK_FLASH_START + sector_num * SECTOR_SIZE;
               ^~~~~~~~~~~~~~~~~~~~~~~
/home/test/zephyr.js/deps/zephyr/subsys/disk/disk_access_flash.c:35:15: note: each undeclared identifier is reported only once for each function it appears in
/home/test/zephyr.js/deps/zephyr/subsys/disk/disk_access_flash.c: In function ‘disk_access_read’:
/home/test/zephyr.js/deps/zephyr/subsys/disk/disk_access_flash.c:76:8: error: ‘CONFIG_DISK_FLASH_MAX_RW_SIZE’ undeclared (first use in this function)
  len = CONFIG_DISK_FLASH_MAX_RW_SIZE;
        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/test/zephyr.js/deps/zephyr/subsys/disk/disk_access_flash.c: In function ‘read_copy_flash_block’:
/home/test/zephyr.js/deps/zephyr/subsys/disk/disk_access_flash.c:107:20: error: ‘CONFIG_DISK_FLASH_ERASE_ALIGNMENT’ undeclared (first use in this function)
  if (start_addr & (CONFIG_DISK_FLASH_ERASE_ALIGNMENT - 1)) {
                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/test/zephyr.js/deps/zephyr/subsys/disk/disk_access_flash.c:114:46: error: ‘CONFIG_DISK_FLASH_MAX_RW_SIZE’ undeclared (first use in this function)
  num_read = GET_NUM_BLOCK(CONFIG_DISK_ERASE_BLOCK_SIZE,
                                              ^~~~~~~~~~                   
/home/test/zephyr.js/deps/zephyr/subsys/disk/disk_access_flash.c: In function ‘update_flash_block’:
/home/test/zephyr.js/deps/zephyr/subsys/disk/disk_access_flash.c:159:60: error: ‘CONFIG_DISK_FLASH_ERASE_ALIGNMENT’ undeclared (first use in this function)
  fl_addr = ROUND_DOWN(start_addr, CONFIG_DISK_FLASH_ERASE_ALIGNMENT);
                                                            ^~~~~~~~~~                       
/home/test/zephyr.js/deps/zephyr/subsys/disk/disk_access_flash.c:169:47: error: ‘CONFIG_DISK_FLASH_MAX_RW_SIZE’ undeclared (first use in this function)
  num_write = GET_NUM_BLOCK(CONFIG_DISK_ERASE_BLOCK_SIZE,
                                               ^~~~~~~~~~                   
/home/test/zephyr.js/deps/zephyr/subsys/disk/disk_access_flash.c: In function ‘disk_access_write’:
/home/test/zephyr.js/deps/zephyr/subsys/disk/disk_access_flash.c:199:17: error: ‘CONFIG_DISK_FLASH_ERASE_ALIGNMENT’ undeclared (first use in this function)
  if (fl_addr & (CONFIG_DISK_FLASH_ERASE_ALIGNMENT - 1)) {
                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/test/zephyr.js/deps/zephyr/subsys/disk/disk_access_flash.c: In function ‘disk_access_ioctl’:
/home/test/zephyr.js/deps/zephyr/subsys/disk/disk_access_flash.c:263:23: error: ‘CONFIG_DISK_VOLUME_SIZE’ undeclared (first use in this function)
   *(uint32_t *)buff = CONFIG_DISK_VOLUME_SIZE / SECTOR_SIZE;
                       ^~~~~~~~~~~~~~~~~~~~~~~
At top level:
/home/test/zephyr.js/deps/zephyr/subsys/disk/disk_access_flash.c:21:16: warning: ‘read_copy_buf’ defined but not used [-Wunused-variable]
 static uint8_t read_copy_buf[CONFIG_DISK_ERASE_BLOCK_SIZE];
                ^~~~~~~~~~~~~
/home/test/zephyr.js/deps/zephyr/scripts/Makefile.build:183: recipe for target 'subsys/disk/disk_access_flash.o' failed
make[5]: *** [subsys/disk/disk_access_flash.o] Error 1
/home/test/zephyr.js/deps/zephyr/scripts/Makefile.build:357: recipe for target 'subsys/disk' failed
make[4]: *** [subsys/disk] Error 2
/home/test/zephyr.js/deps/zephyr/Makefile:928: recipe for target 'subsys' failed
make[3]: *** [subsys] Error 2
make[3]: Leaving directory '/home/test/zephyr.js/outdir/frdm_k64f'
Makefile:173: recipe for target 'sub-make' failed
make[2]: *** [sub-make] Error 2
make[2]: Leaving directory '/home/test/zephyr.js/deps/zephyr'
/home/test/zephyr.js/deps/zephyr/Makefile.inc:82: recipe for target 'all' failed
make[1]: *** [all] Error 2
make[1]: Leaving directory '/home/test/zephyr.js'
Makefile:161: recipe for target 'zephyr' failed
make: *** [zephyr] Error 2

Besides, there is a warning message below.

.config:411:warning: symbol value '' invalid for USB_DW_IRQ_PRI
*
* Restart config...
*
*
* USB
*
USB (USB) [Y/n/?] y
  Designware USB Device Controller Driver (USB_DW) [Y/n/?] y
    DesignWare USB Driver Interrupt priority (USB_DW_IRQ_PRI) [] (NEW)

JammyZhou, 2017-04-24:
After applying the patch below for deps/zephyr, the above error can be fixed.
http://paste.ubuntu.com/24445990/

But new error happened, it looks like USB device support is implemented only for QUARK_SE_C1000 at this moment, while it is required for ashell support.

In file included from /home/test/zephyr.js/deps/zephyr/drivers/usb/device/usb_dc_dw.c:22:0:
/home/test/zephyr.js/deps/zephyr/drivers/usb/device/usb_dw_registers.h:201:2: error: #error "Unsupported board"
 #error "Unsupported board"
  ^~~~~
/home/test/zephyr.js/deps/zephyr/drivers/usb/device/usb_dc_dw.c:23:17: fatal error: clk.h: No such file or directory
 #include "clk.h"
                 ^
compilation terminated.

pfalcon, 2017-04-24:
See also similar report for qemu_x86: https://github.com/01org/zephyr.js/issues/1052

JammyZhou, 2017-04-25:
The patch mentioned in [#1052](https://github.com/01org/zephyr.js/issues/1052) can only fix part of the problem.

brianjjones, 2017-04-26:
Thanks for the info, I'll start looking into it.

brianjjones, 2017-07-06:
It appears there is currently no driver for webusb in Zephyr for any board aside from A101.

brianjjones, 2017-07-06:
@grgustaf Should we lower the priority since this isn't something we can currently fix?

grgustaf, 2017-07-26:
@brianjjones, yeah maybe what we should do is add a list of known blockers preventing ashell from working on other platforms to some documentation file. That way, if someone comes along who really wants it to work on another board they might know what needs to be done in Zephyr.

grgustaf, 2017-07-26:
Hmm, I guess we're being rather imprecise here. It's *IDE* that is dependent on WebUSB, currently, not ashell in particular. We should still be able to make ashell work. And we also talked about getting creative about other ways to connect the IDE - for example, someone suggested it connect with websockets. So this isn't necessary absolutely blocked. But to fully support ashell/ide in the same way that we do for A101, yes we'd need a WebUSB driver I guess.