intel/zephyr.js #1055: [sensor] Missing controller name on sensor and can not set name except default name

cuiyanx, 2017-04-24:
### Description

The sensor controller name is undefined always. And if setting controller as not default name, creating sensor will be failed. This is inconsistent with [/docs/sensors.md](https://github.com/01org/zephyr.js/blob/master/docs/sensors.md).

**Accelerometer Interface:**
```
dictionary AccelerometerOptions : SensorOptions  {
    string controller;       // controller name, default to "bmi160"
    boolean includeGravity;  // not supported currently
};
```
 **GyroscopeSensor Interface:**
```
dictionary GyroscopeOptions : SensorOptions  {
    string controller;  // controller name, default to "bmi160"
};
```
**AmbientLightSensor Interface:**
```
dictionary AmbientLightSensorOptions : SensorOptions  {
    string controller;  // controller name, default to "ADC_0"
    unsigned long pin;  // analog pin where the light is connected
};
```
**TemperatureSensor Interface:**
```
dictionary TemperatureSensorOptions : SensorOptions  {
    string controller;  // controller name, default to "bmi160"
};

```

### Test Code

Test default name:
```
var pins = require("arduino101_pins");

var Frequency = 20;

var Accelerometer = new Accelerometer({
    includeGravity: false,
    frequency: Frequency
});

console.log("Accelerometer controller: ", Accelerometer.controller);
console.log("Accelerometer frequency: ", Accelerometer.frequency);

var Gyroscope = new Gyroscope({
    frequency: Frequency
});

console.log("Gyroscope controller: ", Gyroscope.controller);
console.log("Gyroscope frequency: ", Gyroscope.frequency);

var Temperature = new TemperatureSensor({
    frequency: Frequency
});

console.log("Temperature controller: ", Temperature.controller);
console.log("Temperature frequency: ", Temperature.frequency);

var AmbientLight = new AmbientLightSensor({
    pin: pins.A2
});

console.log("AmbientLight controller: ", AmbientLight.controller);
console.log("AmbientLight pin: ", AmbientLight.pin);
```
[image]

Test controller name with not default name:
```
var Accelerometer = new Accelerometer({
    controller: "AccelerometerSensor",
    includeGravity: false,
    frequency: 20
});

console.log("Accelerometer controller: ", Accelerometer.controller);
console.log("Accelerometer frequency: ", Accelerometer.frequency);
```
[image]

 Test controller name with default name:
```
var Accelerometer = new Accelerometer({
    controller: "bmi160",
    includeGravity: false,
    frequency: 20
});

console.log("Accelerometer controller: ", Accelerometer.controller);
console.log("Accelerometer frequency: ", Accelerometer.frequency);
```
[image]

### Steps to Reproduction

1. Run test default name and test controller name case.
2. Watch the screen and check console log.
 

### Actual Result

The sensor controller name is undefined always and can not set another name.
 

### Expected Result

The sensor controller name is defined and can set any name.
 

### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | 4dda019f | Arduino 101 | April 24, 2017 | Fail

 

### Additional Information

jimmy-huang, 2017-04-28:
@cuiyanx I don't think this is a valid test, as controller is never part of the the Sensor interface, it's a optional argument defined in SensorOptions, which is passed to the constructor.  So it should be always undefined when you look up that property.  Maybe our API document need to updated to make it more clear that if the app doesn't specify a controller name, then on the Arduino, the default controller will be used, for example, the bmi160.  If that make sense.

cuiyanx, 2017-05-03:
@jimmy-huang Yes, this is not the Sensor interface. It is a optional argument for user, but it is a must checking argument for tester. If this argument can be set, it can be defined to any name, not undefined or default name. So, I suggest update API document for making more clear or deleted it for user.

cuiyanx, 2017-07-17:
As I said on #1342, closed

closed 2017-07-17 by cuiyanx