intel/zephyr.js #1064: [net] MPU FAULT occurs when running TCP samples on K64f.

qiaojingx, 2017-04-25:
### Description

MPU FAULT occurs when running TCP samples on K64f.

### Test Code

[TCPClient6.js](https://github.com/01org/zephyr.js/blob/master/samples/TCPClient6.js); [TCPEchoServ6.js](https://github.com/01org/zephyr.js/blob/master/samples/TCPEchoServ6.js)
 
### Steps to Reproduction

1. `make BOARD=frdm_k64f JS=samples/TCPEchoServ6.js`
2. cp outdir/frdm_k64f/zephyr.bin /media/<USERNAME>/MBED/

### Actual Result

MPU FAULT occurs when running TCP samples on K64f.
[image]

 ### Expected Result

Sample run successfully and TCP Server setup.
 
### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | 4dda019 | FRDM-K64f | Apr 25, 2017 | Fail
 
### Additional Information

qiaojingx, 2017-04-25:
[TCPEchoServ6.js](https://github.com/01org/zephyr.js/blob/master/samples/TCPEchoServ6.js) can be setup on Arduino 101 and QEMU. But not work on K64f(#1064) and Linux(#1066).
I am not very clear which platform TCP will support on.
Are [TCPEchoServ6.js](https://github.com/01org/zephyr.js/blob/master/samples/TCPEchoServ6.js) and [TCPClient6.js](https://github.com/01org/zephyr.js/blob/master/samples/TCPClient6.js) both supported on all the platform(A101,K64f,Linux,Qemu)?

pfalcon, 2017-04-25:
TCP would work on the same Zephyr platforms as UDP does, so if UDP samples worked on frdm_k64f, nothing should preclude TCP from working. Getting the right board config may be a (usual) culprit though.

My understanding is that neither "dgram" nor "net" modules are available on Linux though - that would require fair amount of work, and it's better to invest it in Zephyr targets at this stage IMHO. (I'd be amazed if I missed it and Linux support for those modules was implemented ;-) ).

closed 2017-04-28 by grgustaf

qiaojingx, 2017-05-04:
Verified with commit 78bc531. This issue is not reproducible.

PR #1072 (2017-04-28): [net] Fix net module on K64f
Fixes #1064
commit 85493f83 (2017-04-28): [net] Fix net module on K64f (#1072)

Fixes #1064