intel/zephyr.js #1066: [net] can't read net module on linux

qiaojingx, 2017-04-25:
### Description

can't read net module on linux

### Test Code

[TCPClient6.js](https://github.com/01org/zephyr.js/blob/master/samples/TCPClient6.js);
 
### Steps to Reproduction

1. `make BOARD=linux`
2. ./outdir/linux/release/jslinux samples/TCPClient6.js

### Actual Result

[image]

 ### Expected Result

TCP client could run on Linux according to the explanation at the front of [TCPEchoServ6.js](https://github.com/01org/zephyr.js/blob/master/samples/TCPEchoServ6.js)

 
### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | 4dda019 | Linux | Apr 25, 2017 | Fail
 
### Additional Information

jprestwo, 2017-04-25:
@qiaojingx, the "net" module is not supported on Linux, neither is "dgram".

qiaojingx, 2017-04-26:
Closed since net and dgram modules are not supported on Linux.

qiaojingx, 2017-04-26:
Reopen since still have some confusion.

@jprestwo, 
1. How can I run [TCPClient.js](https://github.com/01org/zephyr.js/blob/master/samples/TCPClient6.js) if net module is not supported on linux? Does that means server and client will both run on Board(A101, k64f)?
2. What is the purpose of [TCPClient.js](https://github.com/01org/zephyr.js/blob/master/samples/TCPClient6.js), is that for client testing, or just for cooperating with server testing?
3. As you mentioned in [TCPEchoServ6.js](https://github.com/01org/zephyr.js/blob/master/samples/TCPEchoServ6.js). 
[image]
Does't that mean TCP Client could be set up on Linux? or I am misunderstanding that.

cuiyanx, 2017-04-26:
@jprestwo 
Can I connect socket server with socket client coding by the other programming language(like java or python)?

jprestwo, 2017-04-26:
@cuiyanx,

Sorry for the confusion, I should have mimicked the comments like @pfalcon did with his dgram samples.

1. TCPClient.js is still meant to be run on an Arduino 101 using 6lowpan. I am not sure if you could run it using two arduinos (TCPClient/TCPEchoServ6) but I doubt it because there would be no connection mechanism for BLE.

2. Goes with 1, all the TCP samples are meant for the Arduino/Zephyr board, not linux.

3. That comment means that you have to setup a TCP client on Linux, not related to ZJS at all. You can use `nc` for that:
`echo "hello" | nc -6 <ip> <port>`
This should be mentioned in the sample like it is for dgram.

I do have a python TCP client and server that I used for testing, the reason I didn't push those up is there was really no place for them in our repo. We are addressing this with #1046 and should have a place for it, somewhere like `tests/manual/net/`. For now though if its not too much trouble just finding a TCP client/server for python is the best route. I just found one online and modified it slightly to my usage so I can't push it up due to licensing.

jprestwo, 2017-04-26:
This should help: https://github.com/01org/zephyr.js/pull/1071

qiaojingx, 2017-04-27:
Closed since TCP Client also need to run on Arduino/Zephyr board.

closed 2017-04-27 by qiaojingx