intel/zephyr.js #1067: [reg][ashell] 'AT' is inputted automatically after ashell start up.

qiaojingx, 2017-04-25:
### Description

 'AT' is inputted automatically after ashell start up.

### Test Code

[ashell](https://github.com/01org/zephyr.js/blob/master/docs/ashell.md); [UART.js](https://github.com/01org/zephyr.js/blob/master/samples/UART.js)
 
### Steps to Reproduction

1. `make ashell ROM=256`
2. `make dfu`
3.  `sudo minicom -D /dev/ttyACM0` to connect ashell with terminal

### Actual Result

'AT' is inputted automatically after ashell start up.
[image]
[image]

### Expected Result

No other command is inputted automatically. 
 
### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | 4dda019 | Arduino 101 | Apr 25, 2017 | Fail
master | 0b11671 | Arduino 101 | Apr 18, 2017 | Pass
 
### Additional Information

'AT' will input automatically when test [UART.js](https://github.com/01org/zephyr.js/blob/master/samples/UART.js), too.

grgustaf, 2017-04-26:
AT is the old-timey prefix for modem commands so it seems like something is treating the connection like a modem. Perhaps the instruction to disabled modemmanager wasn't followed (but @brianjjones doesn't seem to think that's the issue here).

brianjjones, 2017-04-26:
I stand corrected, its ModemManager for sure.  Its just that I don't run minicom with sudo, so I didn't get the modem commands.

Anyway, 'killall ModemManager' should fix your issue.  You'll need to do it each time you boot.

grgustaf, 2017-06-07:
@brianjjones, the newer udev rule instructions will also prevent this problem, is that right?

@qiaojingx please verify that one of these solutions works.

qiaojingx, 2017-06-08:
This issue is not reproducible after using newer udev rule. And 'kill all ModemManager' also can prevent this problem.
So closed.
Commit: fee4637

closed 2017-06-08 by qiaojingx