intel/zephyr.js #1070: [ocf] can't read ocf module on Mac

qiaojingx, 2017-04-26:
### Description

can't read ocf module on Mac

### Test Code

[OcfServer.js](https://github.com/01org/zephyr.js/blob/master/samples/OcfServer.js); [OcfPlatformInfo.js](https://github.com/01org/zephyr.js/blob/master/samples/OcfPlatformInfo.js);[OcfNotifyServer.js](https://github.com/01org/zephyr.js/blob/master/samples/OcfNotifyServer.js);[OcfDeviceInfo.js](https://github.com/01org/zephyr.js/blob/master/samples/OcfDeviceInfo.js);[OcfClientUpdate.js](https://github.com/01org/zephyr.js/blob/master/samples/OcfClientUpdate.js);[OcfClient.js](https://github.com/01org/zephyr.js/blob/master/samples/OcfClient.js);
 
### Steps to Reproduction

1. `make BOARD=linux`
2. `./outdir/linux/release/jslinux samples/OcfServer.js `

### Actual Result

[image]

### Expected Result

OcfServer starts
 
### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | 4dda019 | Mac | Apr 26, 2017 | Fail
 
### Additional Information

jprestwo, 2017-04-26:
OCF is not currently supported in Iotivity-constrained, and as such not supported in ZJS. I asked the Iotivity-constrained maintainer and he said that there is a gerrit patch that is in review to add support for Mac. Once that goes in we can start building OCF on Mac.

grgustaf, 2017-06-07:
I think @jprestwo meant to say **MAC** is not supported in iot-c.

qiaojingx, 2017-08-18:
Does OCF plan to be supported on Mac before 0.4 release?

jimmy-huang, 2017-10-26:
It should work with Linux builds as long as iotivity-constrained can be built on Mac, this is the only reason that OCF is not enabled on Mac since upstream iotivity-constrained linux port doesn't build properly on MacOS.