intel/zephyr.js #1099: [timer] Error range is too large on setTimeout function

cuiyanx, 2017-05-04:
### Description

Error range is too large on setTimeout function. 
 

### Test Code

**Simple timer test1:**
```
console.log("Simple timer test1");
var performance = require("performance");

var timer1, timer2, timer3, timer4, timer5, timer6, timer7, timer8;
var begin1, begin2, begin3, begin4, begin5, begin6, begin7, begin8;
var runtime1, runtime2, runtime3, runtime4, runtime5, runtime6, runtime7, runtime8;

console.log("timer", "begin", "runtime", "delay", "error");

begin1 = performance.now();
timer1 = setTimeout(function () {
    runtime1= performance.now();
    console.log("timer1", begin1, runtime1, "3000", runtime1 - begin1);
}, 3000);

begin2 = performance.now();
timer2 = setTimeout(function () {
    runtime2= performance.now();
    console.log("timer2", begin2, runtime2, "4000", runtime2 - begin2);
}, 4000);

begin3 = performance.now();
timer3 = setTimeout(function () {
    runtime3= performance.now();
    console.log("timer3", begin3, runtime3, "5000", runtime3 - begin3);
}, 5000);

begin4 = performance.now();
timer4 = setTimeout(function () {
    runtime4= performance.now();
    console.log("timer4", begin4, runtime4, "6000", runtime4 - begin4);
}, 6000);

begin5 = performance.now();
timer5 = setTimeout(function () {
    runtime5= performance.now();
    console.log("timer5", begin5, runtime5, "7000", runtime5 - begin5);
}, 7000);

begin6 = performance.now();
timer6 = setTimeout(function () {
    runtime6= performance.now();
    console.log("timer6", begin6, runtime6, "8000", runtime6 - begin6);
}, 8000);

begin7 = performance.now();
timer7 = setTimeout(function () {
    runtime7= performance.now();
    console.log("timer7", begin7, runtime7, "9000", runtime7 - begin7);
}, 9000);

begin8 = performance.now();
timer8 = setTimeout(function () {
    runtime8= performance.now();
    console.log("timer8", begin8, runtime8, "10000", runtime8 - begin8);
}, 10000);
```
**Simple timer test2:**
```
console.log("Simple timer test2");
var performance = require("performance");

var timer1, timer2, timer3, timer4, timer5, timer6, timer7, timer8;
var begin1, begin2, begin3, begin4, begin5, begin6, begin7, begin8;
var runtime1, runtime2, runtime3, runtime4, runtime5, runtime6, runtime7, runtime8;

console.log("timer", "begin", "runtime", "delay", "error");

begin1 = performance.now();
timer1 = setTimeout(function () {
    runtime1= performance.now();
    console.log("timer1", begin1, runtime1, "3000", runtime1 - begin1);
}, 3000);

begin2 = performance.now();
timer2 = setTimeout(function () {
    runtime2= performance.now();
    console.log("timer2", begin2, runtime2, "3010", runtime2 - begin2);
}, 3010);

begin3 = performance.now();
timer3 = setTimeout(function () {
    runtime3= performance.now();
    console.log("timer3", begin3, runtime3, "3020", runtime3 - begin3);
}, 3020);

begin4 = performance.now();
timer4 = setTimeout(function () {
    runtime4= performance.now();
    console.log("timer4", begin4, runtime4, "3030", runtime4 - begin4);
}, 3030);

begin5 = performance.now();
timer5 = setTimeout(function () {
    runtime5= performance.now();
    console.log("timer5", begin5, runtime5, "3040", runtime5 - begin5);
}, 3040);

begin6 = performance.now();
timer6 = setTimeout(function () {
    runtime6= performance.now();
    console.log("timer6", begin6, runtime6, "3050", runtime6 - begin6);
}, 3050);

begin7 = performance.now();
timer7 = setTimeout(function () {
    runtime7= performance.now();
    console.log("timer7", begin7, runtime7, "3060", runtime7 - begin7);
}, 3060);

begin8 = performance.now();
timer8 = setTimeout(function () {
    runtime8= performance.now();
    console.log("timer8", begin8, runtime8, "3070", runtime8 - begin8);
}, 3070);
```
**Import performance module in** [/tests/test-pwm.js](https://github.com/01org/zephyr.js/blob/master/tests/test-pwm.js):
Add `console.log(performance.now());` on line 52 and line 54

 

### Steps to Reproduction

1. Run simple timer test1 and check logs.
2. Run simple timer test2 and check logs.
3. Run test-pwm.js and change delay time and run again.
 

### Actual Result

**Simple timer test1:**
[image]

 **Simple timer test2:**
[image]

**test-pwm.js:**

delay | begin | runtime | error
---- | ---- | ---- | ----
3000 | 70 | 3080 | 3010
3010 | 70 | 3140 | 3070
3060 | 70 | 3140 | 3070
3070 | 70 | 3200 | 3130
3090 | 70 | 3200 | 3130

### Expected Result

Work well and accurate delay.
 

### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | 78bc531 | Arduino 101 | May 4, 2017 | Fail

 

### Additional Information

grgustaf, 2017-05-05:
You are probably expecting performance.now() to return an accurate time... as I also expected until I was looking at it a couple months ago. But what it actually does is returns you the same time now matter how long the current JS timeslice has been running. That means if you do a lot of work as you do above, it still shows the same value: 20, 20, 20, 20... but really those operations are slow and time has advanced since you started.

So what I mean is `begin8` is very inaccurate. It may really be 60 by then, but it's still getting set to 20.

My initial thought was to make the time reported more accurate, but there are some security reasons to not get too accurate. Still, we could maybe drop sub-millisecond accuracy but still be right about the milliseconds.

You should get better results in the meantime if you instead chain the timeouts... Set one, then that one sets another, etc.

grgustaf, 2017-05-05:
Also, console.log() is extremely slow. Calling a C API takes about 3000 cycles (Maybe 0.01ms on A101.) But it also takes that long for each character you print. So each console.log here is probably taking about 4ms FWIW.

See [performance experiments](https://github.com/01org/zephyr.js/wiki/Performance-Experiments) for more info.

grgustaf, 2017-05-05:
Hmm if I'm reading it right though there might be a serious bug in that the second timer is happening so much later than the first time in each case. The rest of it seems explainable by the above. @jprestwo, check that out.

cuiyanx, 2017-05-08:
@grgustaf 
Maybe console.log() is extremely slow, or maybe performance.now() is not expected or maybe not. I think that error range can not be ignored because the hardware cycle is nanoseconds or picoseconds. If the problem was caused by console.log() and performance.now(), the test case test-pwm.js will not be failed.

qiaojingx, 2017-05-09:
This issue casues Timers.js can't output result in order on Linux.
[image]
zephyr.js : commit 3950beb
Platform: Utuntu & Mac OS

cuiyanx, 2017-10-12:
Fixed after #1542 merged and closed

closed 2017-10-12 by cuiyanx