intel/zephyr.js #1100: [fs] Error seeking to position

cuiyanx, 2017-05-04:
### Description

Throw out error seeking to position on [/tests/test-fs.js](https://github.com/01org/zephyr.js/blob/master/tests/test-fs.js).
 

### Test Code

[/tests/test-fs.js](https://github.com/01org/zephyr.js/blob/master/tests/test-fs.js)

 

### Steps to Reproduction

Run test-fs.js and check logs.
 

### Actual Result

[image]

 

### Expected Result

Work well
 

### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | 78bc531 | Arduino 101 | May 4, 2017 | Fail

 

### Additional Information

qiaojingx, 2017-05-11:
Verified with commit 011f0fe + patch #1126. This issue is not reproducible.
Will close this issue after patch merged.

closed 2017-05-11 by jimmy-huang

cuiyanx, 2018-01-10:
Fixed and closed

PR #1126 (2017-05-11): [fs] Fix fs.close() API
Fixes #1100
commit b4d13640 (2017-05-11): [fs] Fix fs.close() API (#1126)

Fixes #1100