intel/zephyr.js #1115: [ocf] Can not find resource on OCF client

cuiyanx, 2017-05-09:
### Description

OCFClient can not find resource, it will be hanging.
 

### Test Code

Add console.log on [/samples/OcfClient.js](https://github.com/01org/zephyr.js/blob/master/samples/OcfClient.js):
    Add `console.log("----1----");` on line 44;
    Add `console.log("----2----");` on line 50;

Like this:
```
console.log("----1----");
client.findResources({ resourceType:"core.light" }, onfound).then(function(resource) {
    console.log("findResources() was successful, deviceId=" + resource.deviceId);
}).catch(function(error) {
    console.log("findResources() returned an error: " + error.name);
});
console.log("----2----");
```

Result:
[image]

 

### Steps to Reproduction

Run test case and check console logs.
 

### Actual Result

Can not find resource and will be hanging.
 

### Expected Result

Find resource and work well.
 

### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | 3950beb | Arduino 101 | May 9, 2017 | Fail

 

### Additional Information

qiaojingx, 2017-05-11:
Verified with commit 011f0fe + patch #1125 . This issue is not reproducible.
Will close this issue after patch merged.

closed 2017-05-11 by grgustaf

cuiyanx, 2018-01-10:
Fixed and closed

PR #1125 (2017-05-11): [ocf] Move ocf.start() to correct location in client samples
Fixes #1115
commit 2826a75d (2017-05-11): [ocf] Move ocf.start() to correct location in client samples (#1125)

Fixes #1115