intel/zephyr.js #1121: [ws] new WebSocket() failed in NodeWebSocketClient.js

qiaojingx, 2017-05-09:
### Description

new WebSocket() failed in NodeWebSocketClient.js
 
### Test Code

[NodeWebSocketClient.js](https://github.com/01org/zephyr.js/blob/master/samples/websockets/NodeWebSocketClient.js);
 
Add console.log to test code as below:
[image]

### Steps to Reproduction

1.  `make JS=samples/websockets/NodeWebSocketClient.js`
2.  `make dfu`
### Actual Result

new WebSocket() failed in NodeWebSocketClient.js
On QEMU:
[image]
On Arduino 101:
[image]

### Expected Result

No issue occurs.
 
### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | 3950beb| Arduino 101 | May 9, 2017 | Fail
master | 3950beb| QEMU | May 9, 2017 | Fail
 
### Additional Information

Websocket on K64f is blocked by #1098

jprestwo, 2017-05-09:
I probably should have been more clear via a comment, but `NodeWebSocketClient.js` is a Node.js script, not meant for ZJS. There is currently no web socket client support in ZJS.

qiaojingx, 2017-05-10:
Thanks for explanation. [NodeWebSocketClient.js](https://github.com/01org/zephyr.js/blob/master/samples/websockets/NodeWebSocketClient.js) works by command `node NodeWebSocketClient.js`
So Closed.

closed 2017-05-10 by qiaojingx