intel/zephyr.js #1144: The registered resource type string 'rt' will be truncated to 16 characters

ttzeng, 2017-05-15:
It seems the given `resourceTypes` strings for registering a resource will be truncated. For example, my [led.js server](https://github.com/01org/SmartHome-Demo/blob/1c7b21baf567568c3faf11f0234a8de1d0ad4752/ocf-servers/zjs-servers/led.js#L27) registers a resource of type `oic.r.switch.binary`, but the iot-rest-api-server reports it as a resource of type `oic.r.switch.bin` (see below). Same behavior observed on registering a button resource of type `oic.r.sensor.motion` becomes `oic.r.sensor.mot`, and resource of type `oic.r.temperature` becomes `oic.r.temperatur.`

`{"di":"a4546509-fd54-4509-4d55-65099d556509","links":[{"href":"/a/led","rt":"oic.r.switch.bin","if":"oic.if.baseline","p":{"bm":1,"secure":false}}]}`

Summary:
```
oic.r.switch.binary   --> oic.r.switch.bin
oic.r.sensor.motion --> oic.r.sensor.mot
oic.r.temperature    --> oic.r.temperatur
```

jprestwo, 2017-05-15:
Do you happen to know what the spec says is the maximum length for the "rt" property? I increased it to 32 in #1155 but it really should be changed to what the spec says.

closed 2017-05-15 by grgustaf

ttzeng, 2017-05-16:
@jprestwo , there is no maximum length requirement for the `rt` property neither in the [OCF Core spec](https://www.google.com.tw/url?sa=t&rct=j&q=&esrc=s&source=web&cd=7&cad=rja&uact=8&ved=0ahUKEwi77PewmfPTAhXGErwKHZQzBwsQFgg3MAY&url=https%3A%2F%2Fopenconnectivity.org%2Fspecs%2FOIC_Core_Specification_v1.1.0.pdf&usg=AFQjCNGmXUPZxrrNSspHNKJgv4JCEMjvwA&sig2=Ek1_hO8n_yD3951x9J5-uA) nor [RFC 6690](https://tools.ietf.org/html/rfc6690#section-3.1), though the use of fixed 32 chars `rt` string works for all OCF predefined resource types except the Magnetic Field Direction type (`oic.r.sensor.magneticfielddirection`) in the OCF Resource Type Spec, it would be good not to have `rt` string length limitation in the design in future. Thanks.

jprestwo, 2017-05-16:
Since we target embedded devices we do have to have some upper limit on the length. If 32 is enough, for now at least, lets leave it at that.

commit 90c3140a (2017-05-15): [ocf] Increase resource type string length (#1155)

Fixes #1144