intel/zephyr.js #1148: [reg] Failed to setup IPv6 connection on K64f

qiaojingx, 2017-05-15:
### Description

Failed to setup IPv6 connection on K64f

### Test Code

[TCPEchoServ6.js](https://github.com/01org/zephyr.js/blob/master/samples/TCPEchoServ6.js); [UDPEchoServ6.js](https://github.com/01org/zephyr.js/blob/master/samples/UDPEchoServ6.js);  [OcfServer.js](https://github.com/01org/zephyr.js/blob/master/samples/OcfServer.js); [WebSocketServer.js](https://github.com/01org/zephyr.js/blob/master/samples/websockets/WebSocketServer.js);
 
### Steps to Reproduction

1. `make BOARD=frdm_k64f JS=samples/UDPEchoServ6.js`
2. cp outdir/frdm_k64f/zephyr.bin /media/<USERNAME>/MBED/
3. `echo 'hello' | nc -u -6 2001:db8::1 33333` to sent string to UDPEchoServ6.js

### Actual Result

No string received
 
### Expected Result

String can be received
 
### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | bd07595 | FRDM-K64F | May 15, 2017 | Pass
master | e5f62c0 - e24821c| FRDM-K64F | May 15, 2017 | Fail
 
### Additional Information

The same issue also reproduced when running TCP, WebSocket, OCF.

grgustaf, 2017-06-12:
I don't have the hardware to test this currently; I'd suggest retesting with PR #1231 which may have fixed it.

qiaojingx, 2017-06-19:
Verified with commit 78c9749 which include #1231. But, this issue is still reproducible.
Platform: FRDM-K64F.

qiaojingx, 2017-07-07:
Verified with commit 0334065. This issue is not reproduced. New issue is found for bus fault error in websocket. It is tracking by #1336 .
Platform: FRDM-K64F
ZephyrSDK:0.9.1
Zephyr: 1.8
Close this one.

closed 2017-07-07 by qiaojingx