intel/zephyr.js #1149: [ocf] Callback cannot be called separately on client.findResources function

cuiyanx, 2017-05-15:
### Description

The callback functions always are called together to find different server resources.
 

### Test Code

**Simple client test code with [/samples/OcfMultiSensor.js](https://github.com/01org/zephyr.js/blob/master/samples/OcfMultiSensor.js):**
```
console.log("Test OCF client");

var ocf = require("ocf");

var client = ocf.client;

function onUpdate(resource) {
}

function onError(error) {
}

function onfoundOne(resource) {
    console.log("I am onfoundOne callback");
}

function onfoundTwo(resource) {
    console.log("I am onfoundTwo callback");
}

function onfoundThr(resource) {
    console.log("I am onfoundThr callback");
}

client.on("update", onUpdate);
client.on("error", onError);

ocf.start();

client.findResources({resourceType:"core.light" },
                     onfoundOne).then(function(resource) {
    console.log("----Parting line----");
}).catch(function(error) {
});

client.findResources({resourceType:"core.temp" },
                     onfoundTwo).then(function(resource) {
    console.log("----Parting line----");
}).catch(function(error) {
});

client.findResources({resourceType:"core.light" },
                     onfoundThr).then(function(resource) {
    console.log("----Parting line----");
}).catch(function(error) {
});

```

 

### Steps to Reproduction

Run simple test and check console logs.
 

### Actual Result

[image]

 

### Expected Result

Can be called separately.
 

### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | e24821c | Arduino 101 | May 15, 2017 | Fail

 

### Additional Information

jprestwo, 2017-05-15:
The problem here is the `resourcefound` event is associated with the `client` object so adding multiple listeners will just build up a list and they will all be called at once. This is part of the JS spec we are following. If you want different callbacks you can always use the promise `.then()` function which will be called uniquely for each `resourcefound` event.

grgustaf, 2017-06-07:
@cuiyanx sounds like @jprestwo is saying this is invalid; do you agree?

closed 2017-06-08 by cuiyanx