intel/zephyr.js #1151: [reg][net] Only frist string can be received by TCPEchoServ6.js on QEMU

qiaojingx, 2017-05-15:
### Description

[reg][net] Only first string can be received by TCPEchoServ6.js on QEMU

### Test Code

[TCPEchoServ6.js](https://github.com/01org/zephyr.js/blob/master/samples/TCPEchoServ6.js); 

### Steps to Reproduction

1. `make BOARD=qemu_x86 qemu JS=samples/TCPEchoServ6.js`
2. `nc -6 2001:db8::1 4242` to connect with TCPEchoServ6.js
3. Send string

### Actual Result

Only first string can be received. 
[image]

### Expected Result

String can be received continuously.

### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | bd07595 | QEMU | May 15, 2017 | Pass
master | e5f62c0 - e24821c| QEMU | May 15, 2017 | Fail

### Additional Information

This issue is probably caused by commit e5f62c0.

grgustaf, 2017-06-12:
Fixed by PR #1231.

qiaojingx, 2017-06-14:
Verified with commit 819a41a + #1231. This issue is not reproducible.
Will close it after patch merged.

grgustaf, 2017-06-15:
Qemu was further broken by #1104 but that is fixed by #1247.

closed 2017-06-15 by jimmy-huang

qiaojingx, 2017-06-19:
Verified with commit 78c9749. This issue has been fixed.

commit 04cae5aa (2017-06-15): [net] Update Zephyr and rebase to fix network issues

Disabled OCF tests because iotivity-constrained has not yet been
rebased to work with recent Zephyr builds.

Dropped PME support in default ashell/ide to make images fit again.

Added an include path to checkheaders and fixed a stray typo there.

Fixes #1151.