intel/zephyr.js #1160: [PnP] ARC RAM and ROM sample increase obviously compared with master commit 3950beb.

hanyia, 2017-05-16:
update to SDK 0.9, found a new bug, detail info as below table:

Device | KPIs | ww20 master: e24821c | ww19 master: 3950beb |ww20vs ww19(%)
----|------|----|----|----
Arduino101 | Zephyr_Helloworld_ARC_RAM(Byte)  | 6712 |5592 | 20.03 
Arduino101 | Zephyr_I2C_ARC_ROM(Byte) | 20072 | 16464 | 21.91 
Arduino101 | Zephyr_I2C_ARC_RAM(Byte)  | 6848 | 5728 | 19.55 
Arduino101 | Zephyr_WebBluetoothGroveLcdDemo_ARC_ROM(Byte)  | 24436 | 20220 |20.85
Arduino101 | Zephyr_WebBluetoothGroveLcdDemo_ARC_RAM(Byte)  | 7288 | 6168 | 18.16

jimmy-huang, 2017-05-17:
This is a known issue, because to work-around the bug ZEP-1882 that AIO is broken with 0.9 SDK, we updated Zephyr to use commit that contains the patch to build ARC with -O2 instead of -Os, thus the increased ROM and RAM size.  We should be able to fix this once Zephyr released a fixed SDK.

grgustaf, 2017-06-07:
@jimmy-huang wasn't this fixed later?

hanyia, 2017-07-11:
Verify this issue on latest SDK 0.9.1, only Zephyr_WebBluetoothGroveLcdDemo_ARC_RAM KPI has meet this issue, result is 7184, has 16.47% gap. Other KPIs have fixed.

jimmy-huang, 2017-07-31:
We can probably close this as this is a duplicate of #1367

closed 2017-08-14 by jimmy-huang