intel/zephyr.js #1165: [ashell] String length settings issue between single quotes

qiaojingx, 2017-05-16:
### Description

[ashell] String length settings issue between single quotes
 
### Test Code

[ashell](https://github.com/01org/zephyr.js/blob/master/docs/ashell.md); [Button.js](https://github.com/01org/zephyr.js/blob/master/samples/arduino/digital/Button.js)
 
### Steps to Reproduction

1. `make ashell ROM=256`
2. `make dfu`
3. `acm> load test.js` and Enter
4. `RAW> /*Paste code of Button.js*/`and CTRL+Z to finish transfer
5. `acm> run test.js`

### Actual Result

[ERR] require("...") string is too long
[image]
[image]

 ### Expected Result

Sample runs successfully.

### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | e24821c | Arduino 101 | May 16, 2017 | Fail
 
### Additional Information

If modify [Button.js](https://github.com/01org/zephyr.js/blob/master/samples/arduino/digital/Button.js) as below; this issue will disappear.
Change `var gpio = require('gpio');` to `var gpio = require("gpio");`

haoxli, 2017-05-25:
Verified with commit 82ef9f8 . This issue is not reproducible.

qiaojingx, 2017-05-26:
Closed since not reproduced.

closed 2017-05-26 by qiaojingx