intel/zephyr.js #1173: [net] Can not run server.address() function that will throw out error

cuiyanx, 2017-05-18:
### Description

I test server.address() function after server listening. This function can not run and will throw out error.
 

### Test Code

**Simple test:**

```
var IPv6Address = "2001:db8::1";
var IPv6Port = 9876;
var IPv6Family = 6;

server.listen({port: IPv6Port, host: IPv6Address, family: IPv6Family}, function() {
    console.log("----1----");
    ServerAddressInfo = server.address();
    console.log("----2----");

    console.log("TCP server listen: " + IPv6Address + ":" + IPv6Port);
});
```

 

### Steps to Reproduction

 

### Actual Result

[image]

 

### Expected Result

Get server address information.
 

### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | 8a5bafc| Arduino 101 | May 18, 2017 | Fail

 

### Additional Information

grgustaf, 2017-06-09:
The test code is incomplete; server is not defined there. Are you testing on qemu or what?

grgustaf, 2017-06-09:
I tried to piece it together from other samples and I reproduced the problem in qemu, but in more recent code base it says "Error: Expected a function." which is nicer.

Looking into it, this function is listed in the docs but is not actually implemented currently.

grgustaf, 2017-06-12:
Fixed by PR #1231.

closed 2017-06-15 by jimmy-huang

cuiyanx, 2017-06-19:
@grgustaf I can get server address information, but server address is wrong.

cuiyanx, 2017-07-25:
Created new issue #1353, so closed

commit 53655997 (2017-06-15): [net] Add server.address() function

Fixes #1173.