intel/zephyr.js #1174: [net] Server.getConnections function will be hang when there is socket connection

cuiyanx, 2017-05-18:
### Description

 

### Test Code

Add simple test into [/samples/TCPEchoServ6.js](https://github.com/01org/zephyr.js/blob/master/samples/TCPEchoServ6.js):
```
server.on("connection", function(sock) {
    console.log("----1----");
    server.getConnections(function(count, error) {
        console.log("count:", count);
        console.log("There is one socket connection, I am running");
    });
    console.log("----2----");
});
```

 

### Steps to Reproduction

1. Add simple test into `/samples/TCPEchoServ6.js`
2. Flash into a101
3. Connect to client
4. Check console logs
 

### Actual Result

[image]

 

### Expected Result

Work well.
 

### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | 8a5bafc | Arduino 101 | May 18, 2017 | Fail

 

### Additional Information

grgustaf, 2017-06-12:
Fixed by #1231.

closed 2017-06-15 by jimmy-huang

cuiyanx, 2017-06-19:
@grgustaf I can get connections, but this connections count always is '0'.

cuiyanx, 2017-06-19:
fixed and closed

commit eb13e608 (2017-06-15): [net] Fix two bugs in server_get_connections

The first one made the loop not terminate, the second made the callback
arguments invalid (arg count instead of arg size was passed).

Fixes #1174.