intel/zephyr.js #1187: [net] Throw out an error before the server responds

cuiyanx, 2017-05-22:
### Description

I test socket.connect function as client and find socket connection throw out error so fast, the server hasn't had time to respond yet. So, the socket connection will be failed when first connecting.
 

### Test Code

[/samples/TCPClient6.js](https://github.com/01org/zephyr.js/blob/master/samples/TCPClient6.js)

 

### Steps to Reproduction

 

### Actual Result

If first connecting, socket connection will be failed.
 

### Expected Result

Socket connection wait respondent from server until time out, and throw out error.
 

### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | 82ef9f89c | Arduino 101 | May 22, 2017 | Fail

 

### Additional Information

grgustaf, 2017-08-15:
I cannot seem to reproduce this problem. I tested after applying my patch from #1474.

qiaojingx, 2017-08-18:
Verified with commit 06cce05. It won't go into function `if` in error since error name is not `NotFoundError`. So this issue will not reproduce. But if modify `NotFoundError` to `Error`, still can see this issue.
[image]

grgustaf, 2017-08-29:
OK, I finally see what you mean. First, to help reproduce this in the future, I needed to patch tests/tools/tests-tcp6-server.py to use port 4242 which TCPClient6.js is expected. Then start that on your host, and boot up this system.

The problem is that the first time it shows the "Server not found" message instead of working; it's only on retry that it starts working.

Now that I fully understand the issue we can lower the priority and don't need to fix this for 0.4.

qiaojingx, 2018-01-03:
Verified with commit b32969f on Arduino101, FRDM-K64F and Qemu.
This issue is reproduced contiunously on Arduino101 over Ethernet. But on FRDM-K64F and Qemu, it occurs alternately.

grgustaf, 2018-02-05:
Try to reserve the word `Verified` for when the fix works and the problem is solved. Here you `Tested` it but it still wasn't fixed.