intel/zephyr.js #1188: [net] Socket connection as client will be hang after sending and receiving data 3 times

cuiyanx, 2017-05-22:
### Description

I test socket connection as client and find this socket will be hang after sending and receiving data 3 times. 
 

### Test Code

**Simple test for send and receive data:**
```
socket.on("data", function(buf) {
    console.log("receive data: " + buf.toString("ascii"));

    socket.write(buf, function() {
        console.log("send data: " + buf.toString("ascii"));
    });
});
```

 

### Steps to Reproduction

 

### Actual Result

[image]

 

### Expected Result

Work well.
 

### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master |  	82ef9f8 | Arduino 101 | May 22, 2017 | Fail

 

### Additional Information

grgustaf, 2017-06-12:
This seems to be fixed with #1231, please retest when that goes in. I seem to run into #1151 before I see this.

cuiyanx, 2017-06-19:
@grgustaf 

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | 78c97490a | Arduino 101 | June 19, 2017 | Fail

Log:
```
receive data: Hello
send data: 'hello'
receive data: Hello

src/zjs_net.c:360 socket_write():
(ERROR) cannot acquire send_buf
send data: 'hello'

src/zjs_net.c:381 socket_write():
(ERROR) Cannot send data to peer (-128)
send data: 'hello'

```

grgustaf, 2017-07-24:
I see this very rarely if I use the ENC28J60 Ethernet board w/ A101, but yes it always seems to happen right away with BLE. I will try to look into it further.

grgustaf, 2017-07-26:
Since as far as we know this is board-specific to A101 and specific to using BLE + 6LoWPAN, there's a decent chance this is a bug in the 6LoWPAN implementation which probably hasn't been overly well exercised. Since our networking code sits on top of the Zephyr APIs it's hard to see how this would be the fault of our networking code, although perhaps it has to do with our configuration of BLE etc. So I'm going to lower this to P2 and it may not get dealt with before 0.4.

grgustaf, 2017-07-26:
@cuiyanx it looks like the symptoms reported above changed? Is this bug different from #1261 or the same one?

grgustaf, 2017-07-26:
Didn't mean to close it.

cuiyanx, 2017-07-27:
@grgustaf Yes, currently. So close this one and please pay attention to #1261

closed 2017-07-27 by cuiyanx