intel/zephyr.js #1191: Missing module not displayed in the error message

gvancuts, 2017-05-23:
If the JavaScript app requires a module that is missing, the console displays an error message stating so but does not display what module is missing.

Steps to reproduce this:
1. Run a ZJS image that does *not* include support for `iotivity-constrained`
2. Load [this JavaScript sample](https://github.com/01org/SmartHome-Demo/blob/master/ocf-servers/zjs-servers/led.js) via the Web IDE
3. Run it and the following message will appear on the console:
```
acm> In function require():
NotSupportedError: module not found: ''
```
I would expect the last line to actually display:
```
NotSupportedError: module not found: 'ocf'
```

grgustaf, 2017-06-07:
Hmm, I attempted to fix a very similar problem right before the hackathon; maybe my fix was incomplete or I broke something else at the same time. :)

closed 2017-06-16 by grgustaf

qiaojingx, 2017-06-21:
Verified with commit 78c9749. This issue is not reproducible.
Close this one.

PR #1258 (2017-06-16): [modules] Fix bug where module string printed can be empty
Fixes #1191.
commit d8540893 (2017-06-16): [modules] Fix bug where module string printed can be empty (#1258)

Fixes #1191.