intel/zephyr.js #1193: [ashell] Messy filename display after pasting long code in ashell then reset device

qiaojingx, 2017-05-24:
### Description

If paste long code via `load test.js` command in ashell, it will stuck. Then, reset device and connect to terminal again, it should work fine. But files stored in ashell will display messily after execute `ls`. 
 
### Test Code

[ashell](https://github.com/01org/zephyr.js/blob/master/docs/ashell.md); 
 
### Steps to Reproduction

1. `make ashell ROM=256`
2. `make dfu`
3. Connect to terminal via `minicom -D /dev/ttyACM0`.
4. `acm> load test.js` and Enter
5. `RAW> /* Paste long code (e.g. `[HeartRateDemo.js](https://github.com/01org/zephyr.js/blob/master/samples/HeartRateDemo.js)`) */`
6. reset device and reconnect again.
7. `acm> ls`
### Actual Result

After step 5. it will stuck and no response by CTRL+Z or CTRL+D. This issue is related to #783 
After step 7. Files stored in ashell display messily.
[image]
I have no way to restore it.
 
### Expected Result

After step 5. Not stuck. CTRL+Z or CTRL+D will finish coding in RAW.
After step 7. files stored in ashell display as filename.

### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | 82ef9f8 | Arduino 101 | May 23, 2017 | Fail
 
### Additional Information

grgustaf, 2017-06-07:
@brianjjones, this is worth looking into a bit now just to see how reproducible it is, so that we can be more sure it's really fixed later when we look into it.

I've seen this at least once myself but had no idea how to reproduce it.

qiaojingx, 2017-06-15:
@grgustaf then how can I remove these messy files? They become more and more with the test going on.

brianjjones, 2017-07-18:
The only time I've seen this is when the file system memory on the board gets corrupted.  I have not been able to fix it once a board ends up like this, its a hardware issue not a software one.

Do you see this problem on a different Arduino101?  Its possible there is a software bug that causes this, but I've only seen it when a board went bad on me.

qiaojingx, 2017-07-19:
@brianjjones , two of my Arduino101 reproduced this issue with commit 82ef9f8. But it doesn't occur with commit af7eadd now.
Another question is: how can I delete there garbled files? It seems one of my arduino101's memory is occupied by these files, I can't run FileSystem samples on it.

brianjjones, 2017-08-05:
@qiaojingx  sorry for the late reply.  I haven't figured out how to fix the garbled file issue I'm afraid. But when I saw it, reflashing with a different version didn't fix anything so it was clearly a hardware issue.

jimmy-huang, 2017-08-16:
@qiaojingx I can't reproduce this with master, but I created a sample to erase all files on the file system #1475 , can you see if it helps with removing these files on your Arduino, I can't really test it.

qiaojingx, 2017-08-17:
@jimmy-huang, I have tried with sample FsEraseAll.js, the normal files stored in filesystem can be removed. But the garbled files are still there. Can't remove. :(

jimmy-huang, 2017-08-17:
@qiaojingx Ok, I submitted another patch to wipe the flash using ashell, you can build and run the ide and then in the command console, type erase, and then reboot the device, see if this will fixes your filesystem, #1477

qiaojingx, 2017-08-18:
@jimmy-huang , verified with commit 06cce05c + #1477 . Command `erase` in ide can remove all files include garbled files.
Thanks~

PR #1477 (2017-08-25): [ashell] Added command to erase flash file system
Added ability to erase the flash in Ashell/IDE, this
will help fix corrupted files created on the file system.

Fixes #1193
commit b2bc0ab0 (2017-08-25): [ashell] Added command to erase flash file system (#1477)

Added ability to erase the flash in Ashell/IDE, this
will help fix corrupted files created on the file system.

Fixes #1193
closed 2017-08-18 by qiaojingx