intel/zephyr.js #1194: [ocf] OcfClient.js can't connect to OcfServer

qiaojingx, 2017-05-24:
### Description

OcfClient.js can't find resource on OcfServer. It will wait at `Started OCF client`. But if add `console.log('---1----')` in OcfClient.js. This issue will disappear. We can't locate the root cause. Please help to debug this issue.
 
### Test Code

[OcfClient.js](https://github.com/01org/zephyr.js/blob/master/samples/OcfClient.js);
 
### Steps to Reproduction

1. Setup [OcfServer.js](https://github.com/01org/zephyr.js/blob/master/samples/OcfServer.js) on A101 via BLE according to [linux setup](https://github.com/01org/zephyr.js/blob/master/docs/6lowpan-ble.md#linux-setup)
2. Setup [OcfClient.js](https://github.com/01org/zephyr.js/blob/master/samples/OcfClient.js) on Linux.

### Actual Result

OcfClient.js wait at `Started OCF client`, can't find resource on OcfServer.js.
[image]
 
### Expected Result

OcfClient.js can find resource on OcfServer.js.

### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | 82ef9f8 | Arduino 101 | May 23, 2017 | Fail
master | 82ef9f8 | Linux | May 23, 2017 | Fail
 
### Additional Information

If add `console.log('-----1---------');` in OcfClient.js. This issue will disappear.
[image]

jimmy-huang, 2017-06-08:
@qiaojingx I can't reproduce this problem, when I run OcfClient on Linux, I always get the response, but I get a different error, I testedusing commit 82ef9f89cc345a849b9c8068f4fbed4fbc02a25b and master (fee4637f3ff4c38434a1c06f70cef56a3872495f)

Started OCF server
Registered resource. UUID: d3ef6509-24f0-4509-73f0-6509c2f06509
Connected
oc_network_receive: received 52 bytes
oc_network_receive: incoming message: [fe80:0000:0000:0000:5ef3:70ff:fe78:1d72]:59692
oc_send_buffer: outgoing message: [fe80:0000:0000:0000:5ef3:70ff:fe78:1d72]:59692
oc_send_buffer: failed: (36)

cuiyanx, 2017-06-09:
@jimmy-huang I can reproduce this problem, when I run OcfServer.js on a101 and OcfClient.js on linux.

cuiyanx, 2017-06-12:
@jimmy-huang 
**Server log:**
[image]

**Client log:**
[image]

closed 2017-06-15 by grgustaf

qiaojingx, 2017-06-21:
Verified with commit 78c9749. This issue is not reproducible.
Close this one.

PR #1241 (2017-06-15): [ocf] Fixed OCF Client not able to connect
The promise object needs to be acquired and released later

Fixes #1194
commit 48624ec8 (2017-06-15): [ocf] Fixed OCF Client not able to connect (#1241)

The promise object needs to be acquired and released later

Fixes #1194