intel/zephyr.js #1195: [Docs] ashell.md need to update

qiaojingx, 2017-05-24:
### Description

commands list in ashell has been updated. And `bootcfg FILE   Set the file that should run at boot` and      `set Sets the input mode for 'load' accept data` have been removed. Ashell.md is better to keep consistent.
 ### Test Code

[ashell](https://github.com/01org/zephyr.js/blob/master/docs/ashell.md);
 
### Steps to Reproduction

1. Compare ashell.md and actual output in ashell.

### Actual Result

1. `bootcfg FILE   Set the file that should run at boot` is superfluous in [ashell](https://github.com/01org/zephyr.js/blob/master/docs/ashell.md);

Commands list in ashell.md:
[image]
Commands list in ide:
[image]
 
2. `set Sets the input mode for 'load' accept data` have been removed. So [Data transfer](https://github.com/01org/zephyr.js/blob/master/docs/ashell.md#data-transfer) and [Data transaction example using IHEX](https://github.com/01org/zephyr.js/blob/master/docs/ashell.md#data-transaction-example-using-ihex) are superfluous in [ashell](https://github.com/01org/zephyr.js/blob/master/docs/ashell.md);

### Expected Result

1. `bootcfg FILE   Set the file that should run at boot` should be removed from commands list in [Commands](https://github.com/01org/zephyr.js/blob/master/docs/ashell.md#commands)

2. [Data transfer](https://github.com/01org/zephyr.js/blob/master/docs/ashell.md#data-transfer) and [Data transaction example using IHEX](https://github.com/01org/zephyr.js/blob/master/docs/ashell.md#data-transaction-example-using-ihex) sections should be removed from [ashell](https://github.com/01org/zephyr.js/blob/master/docs/ashell.md);
[image]

### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | 82ef9f8 | Arduino 101 | May 23, 2017 | Fail

### Additional Information

Since commands(`test, error, ping, at, set and get`) have been removed from commands list. Can I close #762, #1028, #1027 directly.

grgustaf, 2017-06-07:
We made this change to clean up the output in the IDE for the hackathon we did two weeks ago. It's okay for some of these commands to continue to exist, and be documented in ashell.md. To update the docs and online help correctly we should consider these categories:

1) Commands the user needs to know about in the IDE console
2) Commands the user needs to know about in serial ashell mode
3) Commands the user might want to know about when they seek more detailed info in the docs.
4) Commands that are intended for machine use only and the user never needs to know about; these should be documented in source but not in ashell.md.
5) Commands that are no longer needed at all and should be removed.

With the patch we made just before the hackathon, we attempted to get category #1 right, that's the new shorter list. But we should think more about the getting the other categories right now.

grgustaf, 2017-06-07:
At the very least I guess we changed `bootcfg` to `boot` so that should be updated.

grgustaf, 2017-08-15:
#1472 synchronizes the ashell docs to match the 'help' output and available commands. Removed the ihex docs as requested. However, the 'boot' command is *not* superfluous; it's useful to be able to set an auto-run script when the device is rebooted.

closed 2017-08-15 by jimmy-huang

qiaojingx, 2017-08-16:
Verified with commit 212747a. This issue is not reproduced.

PR #1472 (2017-08-15): [docs] Update ashell docs to match current command set
Update the help display to match and add documentation for various
commands that weren't documented.

Also, remove mention of ihex transfer because we plan to eventually
remove it entirely, and it's not really useful to the user in ashell
mode.

Fixes #1195
commit 212747a7 (2017-08-15): [docs] Update ashell docs to match current command set (#1472)

Update the help display to match and add documentation for various
commands that weren't documented.

Also, remove mention of ihex transfer because we plan to eventually
remove it entirely, and it's not really useful to the user in ashell
mode.

Fixes #1195