intel/zephyr.js #1197: [ocf] Can not find multiple resources by ocf client

cuiyanx, 2017-05-24:
### Description

The ocf client can not find multiple resources, it just find the last one.
 

### Test Code

Sample as server:
[/samples/OcfMultiSensor.js](https://github.com/01org/zephyr.js/blob/master/samples/OcfMultiSensor.js)

Simple test as client:
```
var ocf = require('ocf');
var client = ocf.client;

ocf.start();

client.findResources({ resourceType:"core.light" }).then(function(resource) {
    console.log("findResources 'core.light' was successful, deviceId=" + resource.deviceId);

    client.retrieve(resource.deviceId, {observable: false}).then(function(res) {
        console.log("light: " + resource.properties.light);
    }, function(error) {
    });
}, function(error) {
    console.log("findResources() returned an error: " + error.name);
});

setTimeout(function() {
    client.findResources({ resourceType:"core.temp" }).then(function(resource) {
        console.log("findResources 'core.temp' was successful, deviceId=" + resource.deviceId);

        client.retrieve(resource.deviceId, {observable: false}).then(function(res) {
            console.log("temp: " + resource.properties.temp);
        }, function(error) {
        });
    }, function(error) {
        console.log("findResources() returned an error: " + error.name);
    });
}, 3000);
```

 

### Steps to Reproduction

1. Build [/samples/OcfMultiSensor.js](https://github.com/01org/zephyr.js/blob/master/samples/OcfMultiSensor.js) on a101
2. Connect with BLE
3. Build and run simple test on linux
4. Check console log
 

### Actual Result

The ocf client just finds the last resource.

**Client:**

[image]

**Server:**

[image]

 

### Expected Result

Can find multiple resources.
 

### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | 82ef9f89c  | Arduino 101 | May 24, 2017 | Fail

 

### Additional Information

jimmy-huang, 2017-07-18:
Can you reproduce this still?  I tried the sample, and there's a slight error, i updated to res.properties.light and res.properties.temp, and I am able to see both resources.

jimmy@jimmy-dev:~/projects/zephyrjs/zephyr.js (fix-ocf-find-resources *)$ outdir/linux/release/jslinux simpletest.js --noexit

call oc_do_ip_discovery
discovery
findResources 'core.light' was successful, deviceId=c7567a09-1557-4a09-6357-7a09b1577a09
light: 4095
call oc_do_ip_discovery
discovery
findResources 'core.temp' was successful, deviceId=c7567a09-1557-4a09-6357-7a09b1577a09
temp: 4095

wanghongjuan, 2017-07-19:
Hi, @jimmy-huang I just tried this sample use commit de91ade, it works.

BTW, API describe function [`findResources(ClientOptions options, optional FoundListener listener);`](https://github.com/01org/zephyr.js/blob/master/docs/ocf.md#clientfindresources): `Find remote resources matching options filter`, so I changed the `OcfMultiSensor.js` and `simpletest.js`, updated all resourceTypes to `resourceTypes:[core.res]`, but only got the latest register temp resource, light is undefined.

cuiyanx, 2017-07-19:
fixed and closed

closed 2017-07-19 by cuiyanx

jimmy-huang, 2017-07-19:
@cuiyanx you can file another bug the the options parameter and we'll track it there.