intel/zephyr.js #1199: Display wrong BLE client address

ttzeng, 2017-05-24:
I use my Nexus 4 to connect to a modified ZJS BLE sample code. The Bluetooth address of my phone should be '98:d6:f7:33:b6:51', but from the 'accept' event callback, it always shows:
```
State change: poweredOn
Advertising start success
Accepted connection from 7b:59:28:25:23:e4 (random)
```

jimmy-huang, 2017-06-06:
this is not the fault of our code, I've asked the Zephyr BLE team before, and they said the underlying Zephyr bluetooth stack is creating a virtual mac address for the client and it doesn't actually return a physical addr unless it's paired.  Since BLE doesn't require pairing, it will always return a virtual addr.

grgustaf, 2017-06-07:
Make sure we explain the above behavior in the relevant API doc. Can we tell whether we have a virtual or physical address? Maybe we could provide an API to let them know if they're looking at a physical or virtual address? Or maybe we know that based on whether we've paired?

jimmy-huang, 2017-06-08:
@grgustaf yeah, the virtual addr ends with this "72:1b:1c:eb:04:33 (random)"