intel/zephyr.js #1203: [f411re] No enough space for flash Nucleo F411RE many times

haoxli, 2017-05-28:
# Description
After flash image to Nucleo F411RE board for many times(>5), it will report that no space left on board, we need to reset the board for next flash.

# Test Code
[HelloWorld.js](https://github.com/01org/zephyr.js/blob/master/samples/HelloWorld.js)

# Steps to Reproduction
1. `$ make JS=samples/HelloWorld.js BOARD=nucleo_f411re`
2. `$ cp outdir/nucleo_f411re/zephyr.bin /media/<username>/NODE_F411RE/`
3. Repeat the 2nd step for more than 5 times

# Actual Result
[image]

`$ minicom -D /dev/ttyACM0`
[image]

# Expected Result
Flash image without error for every times, and execute the code with output 'Hello, ZJS world!'

# Test Builds
Branch | Build ID | Target Device| Date | Result
---|---|---|---|---
master | 6180fec| ST Nucleo F411RE | May 27, 2017 | Fail

grgustaf, 2017-06-07:
This sounds like maybe it's a problem with the board's flashing mechanism, not our code? Could you try this with some native Zephyr images or even non-Zephyr images?

haoxli, 2017-06-08:
Yes, I've tried the native sample application Hello World.
```
$ cd <zephyr_root_path>
$ source zephyr-env.sh
$ cd $ZEPHYR_BASE/samples/hello_world/
$ make BOARD=nucleo_f411re
```
After flash many times, this still work normally.
[image]

grgustaf, 2017-06-08:
Note: we don't have this board but will look into getting one.