intel/zephyr.js #1204: Can't boot BMI160Temperature sample code after upgrading to SDK 0.9.1

ttzeng, 2017-05-31:
After upgrading to Zephyr SDK 0.9.1, the provided temperature sensor sample code doesn't work anymore.
```
make JS=samples/BMI160Temperature.js
```

qiaojingx, 2017-06-01:
This issue can be reproduced with Zephyr SDK 0.9.1. All sensor samples(e.g. [BMI160Temperature.js](https://github.com/01org/zephyr.js/blob/master/samples/BMI160Temperature.js); [BMI160Gyroscope.js](https://github.com/01org/zephyr.js/blob/master/samples/BMI160Gyroscope.js); [BMI160Accelerometer.js](https://github.com/01org/zephyr.js/blob/master/samples/BMI160Accelerometer.js); [AmbientLight.js](https://github.com/01org/zephyr.js/blob/master/samples/AmbientLight.js);) got nothing output in console log.
Platform: Arduino 101
Commit: 6180fec

jimmy-huang, 2017-06-05:
I did some research, it's basically that ADC is broken again after updating to SDK 0.9.1, as the bug https://jira.zephyrproject.org/browse/ZEP-1882 is not yet fixed, so I would not recommend updating to the new 0.9.1 SDK for now.

In the meantime, I pushed a patch to address the issue where accel/gyro/temperature not working with the SDK 0.9.1 SDK as they don't depend on ADC.

https://github.com/01org/zephyr.js/pull/1210

grgustaf, 2017-06-08:
Note from IRC: That JIRA issue has been marked fixed, need to reverify this and make sure there's an open issue w/ Zephyr team if still a problem.

qiaojingx, 2017-07-19:
Verified with commit af7eadd. This issue is not reproducible.

closed 2018-01-10 by jimmy-huang