intel/zephyr.js #1213: [sensor] Sensor stops abnormally

qiaojingx, 2017-06-06:
### Description

When sensor stops, error will be triggered.

### Test Code

[SensorBLEDemo.js](https://github.com/01org/zephyr.js/blob/master/samples/SensorBLEDemo.js); [test-bmi160accelerometer.js](https://github.com/01org/zephyr.js/blob/master/tests/test-bmi160accelerometer.js); [test-bmi160gyroscope.js](https://github.com/01org/zephyr.js/blob/master/tests/test-bmi160gyroscope.js); [test-bmi160temperature.js](https://github.com/01org/zephyr.js/blob/master/tests/test-bmi160temperature.js).
 
### Steps to Reproduction

1.  `make JS=samples/SensorBLEDemo.js ROM=256`
2.  `make dfu`
3.  Connect with nRF app via BLE.
4. Disconnect
### Actual Result

Sensor.stop cause error occurs.
[image]
 
### Expected Result

Sensor stops without error.
 
### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | fee4637 | Arduino 101 | Jun 6, 2017 | Fail
 
### Additional Information

Run auto test cases of sensor. Sensor.stop triggers onerror callbacks.
[image]

uhofemei, 2017-06-08:
I see a similar issue based on PR https://github.com/01org/SmartHome-Demo/pull/150 being that the BLE connection goes down within <= 3 minutes after being established from the command line using the instructions here: https://github.com/01org/zephyr.js/blob/master/docs/6lowpan-ble.md.

qiaojingx, 2017-06-09:
Verified with commit 8d1b343 + #1221. This issue is not reproducible.
Will close after patch merged.

closed 2017-06-15 by grgustaf

qiaojingx, 2017-06-21:
Verified with commit 78c9749. This issue is not reproducible.

PR #1221 (2017-06-15): [sensor] Fixed sensors cannot be stopped in polling mode
Fixes #1213
commit f817ae56 (2017-06-15): [sensor] Fixed sensors cannot be stopped in polling mode (#1221)

Fixes #1213