intel/zephyr.js #1215: [ble] First subscription was unsuccessful with WebBluetooth

qiaojingx, 2017-06-06:
### Description

First subscription was unsuccessful with WebBluetooth(https://goo.gl/QEvyDZ).

### Test Code

[WebBluetoothGroveLcdDemo.js](https://github.com/01org/zephyr.js/blob/master/samples/WebBluetoothGroveLcdDemo.js); 
 
### Steps to Reproduction

1.  `make JS=samples/WebBluetoothGroveLcdDemo.js ROM=256`
2.  `make dfu`
3.  Connect to https://goo.gl/QEvyDZ via BLE.

### Actual Result

JS callback timed out error occurs when first connect.
[image]
 
### Expected Result

No error occurs.
 
### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | fee4637 | Arduino 101 | Jun 6, 2017 | Fail
 
### Additional Information

grgustaf, 2017-06-08:
I think this is an occasional problem, not 100% reproducible, right?

qiaojingx, 2017-06-09:
This problem only reproduced when first connect after A101 reset. It won't occur when second, third connect.
Verified with commit fee4637 + #1224. This issue is fixed. Will closed after patch merged.

closed 2017-06-15 by grgustaf

qiaojingx, 2017-06-21:
Verified with commit 78c9749. This issue is not reproducible.
Close this one.

PR #1224 (2017-06-15): [ble] Fixed WebBluetooth demo fails on initial read
One of the variables needs to be renamed, since it
was actually causing a JS error.

Fixes #1215
commit ae0899e7 (2017-06-15): [ble] Fixed WebBluetooth demo fails on initial read (#1224)

One of the variables needs to be renamed, since it
was actually causing a JS error.

Fixes #1215
PR #1238 (2017-06-15): [ble] Fixed WebBluetooth demo fails on initial read
One of the variables needs to be renamed, since it
was actually causing a JS error.

Fixes #1215
commit a262ad95 (2017-06-15): [ble] Fixed WebBluetooth demo fails on initial read (#1238)

One of the variables needs to be renamed, since it
was actually causing a JS error.

Fixes #1215